merbanan/rtl_433 #308: Inaccurate conversions

ygator, 2016-02-22:
Is there a reason for keeping the inaccurate conversions.  i.e. Temperature 21.1 Celsius is 69.980003 Fahrenheit?  An easy fix is to make all conversion functions in util.c be double instead of float.

merbanan, 2016-02-23:
Can you elaborate why using float is inaccurate?

ygator, 2016-02-23:
Changing to double makes 2.1 C convert exactly to 69.98 F and not 69.980003 F.
For the sensors I have (Lacrosse WS, OS, and OS v1) I change all the floats to doubles and the same for the conversion functions in util.c

merbanan, 2016-02-23:
Ahh ok, then the issue is that float can not represent 69.98 exactly.  I'm not really sure what should be done . When do you need 69.98 instead of 69.980003 ? Is it just for presentation purposes ?

ygator, 2016-02-24:
I just saw that the values are being output with %f.  We can keep the floats if we change this to %.3f or %.2f.

somesortoferror, 2016-02-24:
You have to be a bit careful with conversions that you don't misrepresent the original data.

e.g. 21.1 C is implied uncertainty of +/- 0.05 C
69.98 F is +/- 0.005 F which is much more precise (falsely). Should be +/- 0.09 F

So it's better to round as 70.0 F

(I suspect that many of these 433 MHz devices are closer to +/- 2 C, but there's not much you can do about that...)

ygator, 2016-02-24:
I agree with you, but I think the rounding is better left to the data handler.

merbanan, 2016-03-27:
Did we do something about this issue ? Or is it still valid ?

ygator, 2016-03-27:
Yes, I changed to %f to %.3f.  See #312.  I'll close.

closed 2016-03-27 by ygator