merbanan/rtl_433 #313: Request change in label name "id"

ygator, 2016-02-24:
I'd like to change "id" to something else like "sid" (sensor id).  The reason for this is that the name "id" is usually used as the record id in a database.  In fact I ran across this problem with my setup that uses feathers-sequelize.  I can get around it easy enough by either changing the source or doing a 
`sed 's/\"id\" :/\"sid\" :/'`,  I think it would be better to have it changed globally so that no one else encounters the issue in the future.  I am willing to make all the changes if no one objects.

merbanan, 2016-03-04:
That would break "abi". Can you elaborate why you can't have 2 parameters named id?

ygator, 2016-03-05:
I don't know what "abi" is?  A database cannot have two fields with the same name.  Normally "id" is used as a uniquely identify a record.  In the case of feathers-sequalize it adds this to the record.

merbanan, 2016-03-08:
Ok, I think I will go forward with this type of change but it will be after a first release. And all parameter names will need to be documented and reviewed.

merbanan, 2018-12-03:
As long as you are context aware this is not a problem. Closing issue.

closed 2018-12-03 by merbanan