merbanan/rtl_433 #334: compilation fails in win64

somesortoferror, 2016-03-17:
rtl_433 does not compile successfully under mingw64:
````
/build/mingw64/rtl_433/src/devices/akhan_100F14.c:88:19: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  .demod_arg     = (unsigned long)&pwm_precise_parameters_akhan,
                   ^
/build/mingw64/rtl_433/src/devices/akhan_100F14.c:88:19: error: initializer element is not constant
/build/mingw64/rtl_433/src/devices/akhan_100F14.c:88:19: note: (near initialization for ‘akhan_100F14.demod_arg’)
````
The problem is on Win64 pointers are 64 bit but unsigned longs are 32 bit

A fix is to use uintptr_t instead of unsigned long (defined in C99).

merbanan, 2016-03-24:
Patch welcome. Is this the only instance of this error ? If not a complete log would be nice. I might be able to fix it myself then.

somesortoferror, 2016-03-24:
Submitted. Just wanted to make sure you didn't mind the C99

closed 2016-04-01 by somesortoferror