merbanan/rtl_433 #353: AcuRite Rain Gauge False Positives

ygator, 2016-03-28:
Can someone with an AcuRite Rain Gauge do more processing to filter out it triggering on Oregon Scientific Protocol V1 Sensors.

I am getting:
AcuRite Rain Gauge Total Rain is 1.0mm
Raw Message: 4a 90 02 00 00
{"time" : "2016-03-28 14:54:33", "model" : "OSv1 Temperature Sensor", "sid" : 4, "channel" : 2, "battery" : "OK", "temperature_C" : 19.300}

merbanan, 2016-03-28:
Doesn't look like we have any samples, feel free to disable it.

merbanan, 2016-03-29:
I re thought it. I think what we need is instead a method of disabling a decoder by default. So when we register it the default state is disabled. But it can be enabled with -R34 or what number i got registered at. That way if you have one of these you just enable it explicitly. Does it make sense ?

zuckschwerdt, 2016-03-29:
If I understand correctly the `disabled` parameter in the device struct works that way, i.e. it disables a noisy or development device by default.

merbanan, 2016-03-29:
@zuckschwerdt can you enable it also via -Rx ?

zuckschwerdt, 2016-03-29:
I assumed so and tested it too. Yes, using `-Rx` on a disabled device works as expected.

rct, 2016-03-29:
There are a number of devices that are likely to produce false positives as they have no CRC or checksum.  A number of the generic" remotes" that have been added recently have no error checking. 

IIRC, waveman also generates a lot of false positives, but it's not a big problem for me because I enable the 4 devices I need, but that also goes back to before the performance was vastly improved by @vestom

Should devices without error checking be disabled by default?

merbanan, 2016-03-29:
Devices that are are causing false positives should be disabled by default. Some signals are unique in their own way so they don't need a crc.

rct, 2016-03-29:
If I wasn't running with -R x -R y -R z, I would be of the opinion that Waveman is causing problems.

merbanan, 2016-03-29:
Patch welcome.

rct, 2016-03-29:
I agree that the implementation of the Acurite rain gauge isn't very complete and lacks samples.  

I'm not going to bother to submit a patch for Waveman since I use -R ... -R ...

I'm just trying to note looking forward as more "generic remotes" get implemented using the same chips (EV1527, RT1527, FP1527, HS1527 and HT12E) you might want to consider a class of devices/timings that need to enabled.

closed 2016-03-29 by ygator

homeautomaton, 2023-03-17:
It may be time to revisit this and other "disabled" protocols. There's now a priority mechanism which seems to solve (at least) the conflict between AcuRite Rain Gauge and Oregon V1, if I do this:

```
--- a/src/devices/acurite.c
+++ b/src/devices/acurite.c
@@ -1876,7 +1876,7 @@ r_device const acurite_rain_896 = {
         .gap_limit   = 3500,
         .reset_limit = 5000,
         .decode_fn   = &acurite_rain_896_decode,
-        .disabled    = 1, // Disabled by default due to false positives on oregon scientific v1 protocol see issue #353
+        .priority    = 10, // Eliminate false positives by letting oregon scientific v1 protocol go earlier
         .fields      = acurite_rain_gauge_output_fields,
 };
 ```