merbanan/rtl_433 #356: How to print "logstyle" format?

SledgehammerPL, 2016-03-29:
2016-03-29 22:49:23 :   OSv1 Temperature Sensor
        SID:     7
        Channel:         1
        Battery:         OK
        Temperature:     7.3 C

is very hard to decode, how to get 

2016-03-29 22:49:23: OSv1 Temperature Sensor, SID: 7, Channel: 1, Battery: OK, Temperature: 7.3 C

ygator, 2016-03-29:
rtl_433 -R 49 -F csv
time,model,id,channel,battery,temperature_C
2016-03-29 18:21:39,OSv1 Temperature Sensor,,1,OK,21.700
2016-03-29 18:21:39,OSv1 Temperature Sensor,,1,OK,21.700
2016-03-29 18:21:46,OSv1 Temperature Sensor,,2,OK,20.200

SledgehammerPL, 2016-03-29:
Not exactly - I have more than one sensor with different results: Energy sensor, temperature sensor, and temperature with humidity sensor. So columns don't match, do they?

SledgehammerPL, 2016-03-29:
2016-03-30 01:36:53,OSv1 Temperature Sensor,,1,OK,6.400,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
2016-03-30 01:36:53,OSv1 Temperature Sensor,,1,OK,6.400,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
2016-03-30 01:37:23,OSv1 Temperature Sensor,,1,OK,6.300,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
2016-03-30 01:37:23,OSv1 Temperature Sensor,,1,OK,6.300,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
2016-03-30 01:37:32,Energy Sensor CM160 Current,26,,OK,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,

                                                                 no current no power values

SledgehammerPL, 2016-03-29:
Ok, I found, that I have to rename power to power0 or add power to csv attr list.

closed 2016-03-29 by SledgehammerPL