merbanan/rtl_433 #364: Insight on the AcuRite 00606TX Temperature Only Tower

czenzel, 2016-04-13:
Hello everyone,

I am sorry if I am posting this in the wrong place. I found an AcuRite 00606 TX Temperature Only Tower being used and decoded part of the data.

I have seen in the tests set that someone also uploaded sample data, but the 00606 TX is not implemented. The rtl_433 program is also not detecting is as a temperature sensor from AcuRite.

Here is what I have found:

The first two hexadecimal values are always the same in a transmission. The hex values are ``0d 80``. The next two values are different each time; however, I found they are the temperature (and maybe dew point) in Celsius.

For example right now I am reading ``9f 88`` which equals values of:

Temperature: 15.9 C (0x9F / 10.0f)  

Dew Point ?: 13.6 C (0x88 / 10.0f) --- this value I am unsure of, but looks close to correct.

I was able to verify the temperature value with the indoor display I have. I don't have a display of the dew point being broadcast.

Can somebody confirm? I hope this helps with a possible implementation of the sensor.

Thanks,

Chris

khkremer, 2016-05-14:
The first first byte is actually not always the same: It's a random code that resets when the sensor is reset (e.g. when you change the battery), the second byte contains the "battery OK" flag in it's MSB, all other bits seem to be zero. Then you have two bytes with the temperature, plus a checksum. 

See here for more information: https://forum.arduino.cc/index.php?topic=352986.0

I had an ancient version of rt._433 working with this sensor, but it does not compile anymore with anything recent. I need to invest some time to get this working again.

merbanan, 2018-12-05:
No activity and no references samples. Closing issue.

closed 2018-12-05 by merbanan