merbanan/rtl_433 #370: Specific vs Generic guidance

PhantomSage, 2016-05-03:
Hi,
I have a bunch of sensors based on EV1527, which could be considered a very generic device.
However, the decoder "Akhan 100F14 remote keyless entry" "captures" the signal and says "The data received is not used by the akham keyfob. This might be another device using a HS1527 OTP encoder with the same timing."
To make it even more overlapping, Akhan "mute/0x04" signal, is identical (different ID) to arm on my keyfob.

Is there some guidance on how to write decoders that overlap?
Perhaps there should be a "enable" application layer decoding flag, to enable high level decoders in favor of low level generic chip decoders?
I'm considering hacking away a ev1527 & a kerui security system decoders, but they WILL overlap.
Of course I can "nuke" akhan during compile time, or activated devices, but some generic methodology would be good to have.

merbanan, 2016-05-09:
Can you post output from your sensor ? And there is a flag that needs an forced enable to be used. Try .disabled       = 1,

merbanan, 2018-12-03:
The flex decoder should be used for devices lite this. Closing issue.

closed 2018-12-03 by merbanan