merbanan/rtl_433 #371: Support for "Hideki Wind sensor"

Matthies, 2016-05-07:
Hi.

I own a weather station TE827NL https://www.conrad.de/de/funk-wetterstation-te827nl-672147.html
The Termo-/Hydrosensor is recognized by rtl_433 as "Hideki TS04". The wind sensor of the weather station isn't recognized out of the box. It delivers data for
- temperature
- wind strength
- wind direction

I have patched pulse_demod_clock_bits() and hideki.c to get these values from the wind sensor but before I send you this patch I would like to know if it is okay to "extend" the existing hideki device by adding the output fields for the wind (this is the way I patched hideki.c) or if I should introduce a new device (adding some r_device hideki_wind which would be almost identical to r_device hideki_ts04 but offers different output fields). Device selection can be done by counting the valid bytes in the callback (the wind sensor sends 14 bytes instead of 10). The protocol itself is the same.

Andreas

PS: My rain sensor which also comes with the weather station is broken: Sends data but doesn't count any rain so I cannot analyze the protocol data.

merbanan, 2016-05-07:
Add to hideki.c and send the patch.

Matthies, 2016-05-08:
Heres the patch. The change in pulse_demod_clock_bits() was necessary because the wind sensor (and also the rain sensor) doesn't send a short gap at the very last bit of the message. I hope this doesn't break other sensors that use this function.

[hideki-wind.diff.zip](https://github.com/merbanan/rtl_433/files/253910/hideki-wind.diff.zip)

merbanan, 2016-05-20:
Can you send this a proper pull request and also some sample files ?

guelinator, 2016-07-15:
What happened to the hideki wind sensor support. I am also seeking for this, but it seems as soon as i apply this patch, the rain sensor wont work.

merbanan, 2016-07-15:
No github PR and no signal samples. Without that I can't merge the support.

ludwich66, 2016-10-25:
I added 3 Rainsensorsamples today
ludwich

ludwich66, 2017-03-18:
This issue is done
please close it

skx, 2018-03-10:
@Matthies - Can this be closed?  @ludwich66 suggests support should now be enabled/present...

merbanan, 2018-03-11:
By looking at the hideki-wind.diff compared to the current source it looks like the support is there. Thus closing.

closed 2018-03-11 by merbanan

Matthies, 2018-03-12:
Sorry for late answer. Yes, can be closed.