merbanan/rtl_433 #373: Add brand field in data fields

dducret, 2016-05-10:
For facilitating decoding it would be nice to have a brand field ie for Oregon Scientific:

`           "time",       "",           DATA_STRING, time_str,
            "brand",      "",           DATA_STRING, "OS",
            "model",      "",           DATA_STRING, "WGR968",`

Due to too many changes not tested on others files I post attached the "cleaned" version of oregon_scientific.c than to pull it.

[oregon_scientific.c.txt](https://github.com/merbanan/rtl_433/files/256776/oregon_scientific.c.txt)

merbanan, 2016-05-10:
Can you open a github git pull request ? I have a hard time to review the text file you sent me.

dducret, 2016-05-10:
Done.

merbanan, 2016-05-20:
I will close this. Eventually I will have the motivation to commit the patch you sent as a PR.

closed 2016-05-20 by merbanan

rct, 2016-05-20:
FWIW, Looking down the road and assuming continued growth in decoded devices, I've felt that it would be helpful to have "families" and "devices" instead of having to rely on the device field for everything ie 'oregon', 'acurite', 'lacrosse', ...     is this something you'd consider?