merbanan/rtl_433 #388: Group bit not used for Proove

gnbl, 2016-06-05:
I have an REV Telecontrol 008341C-1 remote with 4x2 buttons 1,2,3,G on/off that looks identical to the ones depicted on http://tech.jolowe.se/home-automation-rf-protocol-update/ , where they are called "Proove from 'Kjell & Company', Anslut from 'Jula',". The device sends 6 packets. The G button transmits the same as the 1 button but has the group bit set. This is not output by rtl_433, although the bits are identified in the decoder header
  https://github.com/merbanan/rtl_433/blob/master/src/devices/proove.c#L11
and are also extracted
  https://github.com/merbanan/rtl_433/blob/master/src/devices/proove.c#L54

I guess it just needs to make it into data
https://github.com/merbanan/rtl_433/blob/master/src/devices/proove.c#L62
and *output_fields[]
https://github.com/merbanan/rtl_433/blob/master/src/devices/proove.c#L75

I used the 64 bit build from 
http://cognito.me.uk/computers/rtl_433-windows-binary-32-bit/
http://cognito.me.uk/uploads/rtl433/rtl433_win64_13042016.zip

Thanks for your efforts!

merbanan, 2018-12-08:
Changes added to the code. Untested but looked trivial.

closed 2018-12-08 by merbanan

gnbl, 2018-12-08:
I'm not sure that I even have these anymore. If I should find them, I shall remember this and test... :-)