merbanan/rtl_433 #390: empty bitbuffer / adding new device (blyss doorbell)

johnjore, 2016-06-17:
Trying to add support for a blyss doorbell (DC5-UK-WH) device. Having read a number of posts on how to calculate short/long/reset limit, I'm currently using this:
```
r_device blyss = {
	.name           = "blyss DC5-UK-WH (433.92 MHz)",
	.modulation     = OOK_PULSE_PWM_RAW,
	.short_limit    = 250,		//p_limit
	.long_limit     = 1800,		//packet distance
	.reset_limit    = 10000,	
	.json_callback  = &blyss_callback,
	.disabled       = 0,
	.demod_arg      = 0,
};

```
when running in -a mode, I see this:

> Exact sample rate is: 250000.000414 Hz
> Sample rate set to 250000.
> Bit detection level set to 8000.
> Tuner gain set to Auto.
> Reading samples in async mode...
> Tuned to 433920000 Hz.
> *** signal_start = 668968, signal_end = 956733
> signal_len = 287765,  pulses = 495
> Iteration 1. t: 250    min: 126 (255)    max: 375 (240)    delta 13
> Iteration 2. t: 250    min: 126 (255)    max: 375 (240)    delta 0
> Pulse coding: Short pulse length 126 - Long pulse length 375
> 
> Short distance: 124, long distance: 373, packet distance: 1743
> 
> p_limit: 250
> bitbuffer:: Number of rows: 16
> [00] {1} 00 : 0
> [01] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [02] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [03] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [04] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [05] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [06] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [07] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [08] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [09] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [10] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [11] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [12] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [13] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [14] {33} 31 71 d5 93 00 : 00110001 01110001 11010101 10010011 0
> [15] {32} 31 71 d5 93 : 00110001 01110001 11010101 10010011

When I run without -a, I get this:

> Exact sample rate is: 250000.000414 Hz
> Sample rate set to 250000.
> Bit detection level set to 8000.
> Tuner gain set to Auto.
> Reading samples in async mode...
> Tuned to 433920000 Hz.
> blyss callback is running!
> blyss DC5-UK-WH:
> bitbuffer:: Number of rows: 16
> [00] {1} 00 : 0
> [01] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [02] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [03] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [04] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [05] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [06] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [07] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [08] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [09] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [10] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [11] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [12] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [13] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [14] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
> [15] {32} 00 00 00 00 : 00000000 00000000 00000000 00000000

My callback is running, it has the same number of rows, and bit lenght is the same, but no actuall 'bits' ?!?
```
			fprintf(stdout, "blyss DC5-UK-WH:\n");
			bitbuffer_print(bitbuffer);
```

I thought -a by default used OOK_PULSE_PWM_RAW, and I was expecting to see the same data. Hopefully its something simple I've done wrong, but a pointer in the right direction would be helpfull as can't figure out what!

JJ

merbanan, 2016-06-17:
Wrong modulation, use PPM. And try with -A also.

johnjore, 2016-06-17:
Ha! Thanks for  this!

I initially got all FF's as the values,  so I switched back to PWM, but:

with -A
```
Detected OOK package
Analyzing pulses...
Total count:  495,  width: 267871               (1071.5 ms)
Pulse width distribution:
 [ 0] count:  255,  width:   128 [127;131]      ( 512 us)
 [ 1] count:  240,  width:   377 [375;380]      (1508 us)
Gap width distribution:
 [ 0] count:   15,  width:  1742 [1741;1744]    (6968 us)
 [ 1] count:  240,  width:   371 [369;373]      (1484 us)
 [ 2] count:  239,  width:   122 [119;125]      ( 488 us)
Pulse period distribution:
 [ 0] count:   15,  width:  1871 [1870;1873]    (7484 us)
 [ 1] count:  479,  width:   499 [498;503]      (1996 us)
Level estimates [high, low]:  15874,    801
Frequency offsets [F1, F2]:   19193,      0     (+73.2 kHz, +0.0 kHz)
Guessing modulation: No clue...

Detected OOK package
Analyzing pulses...
Total count:  495,  width: 267871               (1071.5 ms)
Pulse width distribution:
 [ 0] count:  300,  width:   128 [126;131]      ( 512 us)
 [ 1] count:  195,  width:   377 [376;380]      (1508 us)
Gap width distribution:
 [ 0] count:   15,  width:  1742 [1741;1744]    (6968 us)
 [ 1] count:  285,  width:   371 [369;373]      (1484 us)
 [ 2] count:  194,  width:   122 [119;125]      ( 488 us)
Pulse period distribution:
 [ 0] count:   15,  width:  1871 [1870;1872]    (7484 us)
 [ 1] count:  479,  width:   499 [497;503]      (1996 us)
Level estimates [high, low]:  15889,    427
Frequency offsets [F1, F2]:   19410,      0     (+74.0 kHz, +0.0 kHz)
Guessing modulation: No clue...
```

Using
```
 	.modulation     = OOK_PULSE_PPM_RAW,
 	.short_limit    = 1250,		//p_limit
 	.long_limit     = 2000,		//packet distance
	.reset_limit    = 10000,	
```

I now see:
> bitbuffer:: Number of rows: 16
> [00] {0} :
> [01] {32} e7 37 7a 2c : 11100111 00110111 01111010 00101100
> [02] {32} e7 37 7a 2c : 11100111 00110111 01111010 00101100
> [03] {32} e7 37 7a 2c : 11100111 00110111 01111010 00101100
> 

As this is a door bell, the values don't really have any meaning to me. Can you suggest better values for short/long?

JJ

merbanan, 2016-06-17:
Pulse width distribution:
 [ 0] count:  300,  width:   128 [126;131]      ( 512 us)
 [ 1] count:  195,  width:   377 [376;380]      (1508 us)

Here we have 2 types of pulses. One 512us long and one 1508 us. So we have pulse width modulation. I was wrong before. Set the short limit to (long pulse + short pulse) / 2. That should give good bit detection. 

Gap width distribution:
 [ 0] count:   15,  width:  1742 [1741;1744]    (6968 us)
The message repeat pause should be set to half this value or so, so set long_limit to 4000. And then set reset_limit to 10000.

johnjore, 2016-06-17:
I have much to learn in this area. Thanks for your help, flipped it back to PWM and with the updated short/long, and I see data!

JJ

closed 2016-06-17 by johnjore