merbanan/rtl_433 #393: FSK & Manchester - how?

johnjore, 2016-07-02:
Long story short, is it possible for the maintainers to add modulation type "FSK_PULSE_MANCHESTER" using 'pulse_demod_manchester_zerobit()' ?

Long story:
Looking at the existing code, pulse_detect.c, I can see the "hook" for OSK and Manchester encoding:

> } else if(hist_pulses.bins_count == 2 && hist_gaps.bins_count == 2 && hist_periods.bins_count == 3) {
> 		fprintf(stderr, "**Manchester coding**\n");
> 		device.modulation	= **OOK_PULSE_MANCHESTER_ZEROBIT**;
> 		device.short_limit	= min(hist_pulses.bins[0].mean, hist_pulses.bins[1].mean);		// Assume shortest pulse is half period
> 		device.long_limit	= 0; // Not used
> 		device.reset_limit	= hist_gaps.bins[hist_gaps.bins_count-1].max + 1;				// Set limit above biggest gap
> 	} else if((hist_pulses.bins_count >= 3 && hist_gaps.bins_count >= 3)

but nothing for FSK and Manchester encoding. The debug output:

> Detected FSK package
> Analyzing pulses...
> Total count:   62,  width:  7340                (29.4 ms)
> Pulse width distribution:
>  [ 0] count:   52,  width:    46 [43;52]        ( 184 us)
>  [ 1] count:   10,  width:    97 [97;98]        ( 388 us)
> Gap width distribution:
>  [ 0] count:   51,  width:    56 [51;58]        ( 224 us)
>  [ 1] count:   10,  width:   107 [103;109]      ( 428 us)
> Pulse period distribution:
>  [ 0] count:   43,  width:   102 [97;109]       ( 408 us)
>  [ 1] count:   16,  width:   153 [149;155]      ( 612 us)
>  [ 2] count:    2,  width:   205 [205;206]      ( 820 us)
> Level estimates [high, low]:  12510,    112
> Frequency offsets [F1, F2]:    1707,  -6496     (+6.5 kHz, -24.8 kHz)
> Guessing modulation: Manchester coding
> Attempting demodulation... short_limit: 46, long_limit: 0, reset_limit: 110, demod_arg: 0
> pulse_demod_manchester_zerobit(): Analyzer Device
> bitbuffer:: Number of rows: 1
> **[00] {72} 00 00 7f ed e6 1b 93 00 43**

decodes the data in a way I can interpret. (last line in bold). I'm assuming its not possible to "convert" the bitbuffer passed to the callback function from one type to another, but a "FSK_PULSE_MANCHESTER" "type" needs to be added?

I tried to add the type, but it quickly went very wrong, C was something I used about 20 years ago.... As above the debug output I'm looking for is using pulse_demod_manchester_zerobit().

This is what a dump looks like when using      ".modulation     = FSK_PULSE_PCM," in the device configuration:

> Detected FSK package
> infineon TPMS - start
> bitbuffer:: Number of rows: 1
> [00] {640} f0 f0 78 78 3c 1e 0f 07 83 c1 e0 f0 78 3c 1e 0f 07 80 3f c1 e0 f0 78 3c 1e 0f 0f 07 83 c0 1f e0 f0 07 f8 3c 1e 0f 00 78 3f c1 e0 0f 07 87 83 fc 1e 00 ff 07 83 c0 1e 0f f0 03 c1 fe 0f 00 78 3c 1e 0f 07 83 c1 e1 e0 ff 00 78 3c 1e 0f f0 78 00
Even with a vivid imagination, I can't see how this can end up looking like the analyzers output?

JJ

(I've managed to extract the pre-amble, sensor ID's, temperature and pressure from the string above using C#. Two bytes are unknown, no match on CRC's yet and battery needs a bit more work. I _could_ run rtl in debug and Analyzer mode and capture the strings as they fly by, but a bit ugly)

merbanan, 2016-07-03:
Ok, it  looks like the api design could use a split on the modulation and bit coding. Anyway I try to find some time to decode this signal a little bit.

merbanan, 2016-07-03:
Running rtl_433 -W -m 2 -r ../../rtl_433_tests/tests/steelmate-TPMS/02/gfile001.data out.raw and looking at the output with audacity I clearly see that the signal is FSK modulated and it uses a manchester coding.

merbanan, 2016-07-03:
The new samples are are better but now the signal amplitude is so strong that sometimes it is clipping.

merbanan, 2016-07-03:
steelmate-TPMS/01/gfile001.data Actually looks fine after the fsk demodulation. The trick now is output a nice bit stream.

merbanan, 2016-07-03:
Ok, I think we have to decode it the same way as the oil_watchman.c code. First FSK_PULSE_PCM demodulation. Then in the decoder we take care of the manchester encoding.

johnjore, 2016-07-03:
Hi and thanks for looking at the samples. The issue is what values to use in the definition for the device. Based on your findings, what values should I use for short/long/reset?
(I tend to get gibberish, or nothing, probably because I dont know how to interpret the output from -a and -A. I **almost** good values when using 46 but there are a few errors, the -A gets it correctly, when replaying a specific file:

```
r_device infineon = {
    .name           = "infineon TPMS",
    .modulation     = FSK_PULSE_PCM,
    .short_limit    = 46
    .long_limit     = 46
    .reset_limit    = 4000,
    .json_callback  = &infineon_callback,
    .disabled       = 0,
};
```

I get this:
> bitbuffer:: Number of rows: 1
> [00] {145} aa aa aa aa 6a aa a9 a6 a9 69 56 9a 96 5a 55 55 65 5a 00
Number of rows: 1
**> [00] {72} 00 00 80 12 19 e4 6c ff bc**
Line above is wrong:

But this is the correctly decoded
> Attempting demodulation... short_limit: 46, long_limit: 0, reset_limit: 110, demod_arg: 0
> pulse_demod_manchester_zerobit(): Analyzer Device
> bitbuffer:: Number of rows: 1
> [00] {72} 00 00 7f ed e6 1b 93 00 43
Libe above is correct:

merbanan, 2016-07-03:
git pull and then run:

rtl_433 -D -r ../../rtl_433_tests/tests/steelmate-TPMS/01/gfile003.data

Test mode active. Reading samples from file: ../../rtl_433_tests/tests/steelmate-TPMS/01/gfile003.data
Input format: uint8
Steelmate TPMS decoder
bitbuffer:: Number of rows: 1 
[00] {72} 00 00 7f 45 ea 8d 75 2e f3 
Steelmate TPMS decoder
bitbuffer:: Number of rows: 1 
[00] {72} 00 00 7f 45 ea 8d 75 2e f3 
Steelmate TPMS decoder
bitbuffer:: Number of rows: 1 
[00] {72} 00 00 7f 45 ea 8d 75 2e f3 
Test mode file issued 1 packets

Can you take it from there ?

johnjore, 2016-07-03:
Yes & thanks!

> Registering protocol "Steelmate TPMS"
> Test mode active. Reading samples from file: gfile011.data
> Input format: uint8
> {"time" : "@0.000000s", "type" : "TPMS", "make" : "Steelmate", "id" : 46951, "pressurePSI" : 19.500, "temperature_C" : 12.222, "unknown_1" : -91, "unknown_2" : -29}
> Test mode file issued 1 packets

Needs a few minor fixes, but its pretty much there now :)

closed 2016-07-03 by johnjore