merbanan/rtl_433 #404: Compilation issue

midnightwarrior, 2016-08-04:
Hello!

When I try to build the latest revision of rtl_433 (commit e2f70d1), it fails with this error:
```
rtl_433.o: In function `main':
/opt/rtl/rtl_433/src/rtl_433.c:844: undefined reference to `lightwave_rf'
/opt/rtl/rtl_433/src/rtl_433.c:844: undefined reference to `elro_db286a'
/opt/rtl/rtl_433/src/rtl_433.c:844: undefined reference to `efergy_optical'
/opt/rtl/rtl_433/src/rtl_433.c:844: undefined reference to `hondaremote'
```
I have successfully compiled rtl_433 before on another machine, but this doesn't seem to work.  Is there something I'm doing wrong?

midnightwarrior, 2016-08-04:
I've edited include/rtl_433_devices.h to remove lightwave_rf and the other offending entries from the DEVICES #include, and it compiles now.

merbanan, 2016-08-06:
Not reproducible with current git. I am building with cmake.

closed 2016-08-06 by merbanan