merbanan/rtl_433 #406: Clean up fprintf() usage + auto enable

merbanan, 2016-08-06:
The code modules under devices/*.c needs a review. All fprintf() output should be to stderr and no fprintf() should be on their own. All fprintf()'s must be under debug output guards.

Some modules are not yet converted to make_data() format. Set the enable flag for those modules so they need to be explicitly enabled via the -R parameter.

Remove all printf() usage.

rct, 2016-08-31:
> _"no fprintf() should be on their own. All fprintf()'s must be under debug output guards."_ 

This seems a bit incomplete and/or heavy handed -  You're conflating debug output and error reporting.   There are exceptions that should be reported that aren't necessarily structured data and aren't debugging output.

How should exceptions be reported?

How should error reporting be done (when it's desired)?

rct, 2016-08-31:
See #419 - I think part of  this issue is to have an easy way to get cleanly parseable structured data output.  While clean up is still necessary, rather than just disable all of the unstructured devices, I think a better way is to start identifying which devices output clean structured data, in a manner that can be used by command line option processing.

ludwich66, 2017-01-04:
Please see my google groups entry.
https://groups.google.com/forum/#!topic/rtl_433/yogNIYaHZOM

merbanan, 2018-12-03:
By default there should be no printfs now. Some old decoders might use printf but they are disabled by default. Closing issue.

closed 2018-12-03 by merbanan