merbanan/rtl_433 #413: Honda Remote info

planetclegg, 2016-08-10:
Just noticed "Honda Remote" was added, any info on which Honda models/remotes work?     my 1999 honda remote doesn't appear to be recognized (although I can see what looks like raw data in -a mode)

merbanan, 2016-08-10:
Ask the author on the mailing list. I have no further info. You could add your own remote if you send sample signals.

closed 2016-08-10 by merbanan

ukshady, 2016-08-14:
It's for a 2004 Honda Accord, but if you record some samples and upload to rtl_433_tests I could see if could be modified.

planetclegg, 2016-08-15:
I'll look into it, if I figure out how recording works, I may give a try at decoding it myself.   I already have a small pull request for a different sensor, we'll see how it goes.   It occurs to me the format may change every few model years, may get tricky to lump all of them into one 'device'