merbanan/rtl_433 #416: CRC

merbanan, 2016-08-27:
Add CRC = OK for decoders that support crc.

enavarro222, 2016-09-04:
Did you mean as a "CRC" output field or within device configuration struct ?

merbanan, 2016-09-04:
In the json output. Consumers should be able to trust crc enabled devices more then devices without crc.

rct, 2016-09-18:
If devices drop messages that don't have a valid CRC, what purpose would this serve?

merbanan, 2016-09-18:
Data consumers can add trust to the messages. For example a temperature sensor that transmits a crc has equal trust as one without crc if we don't add a crc tag. The idea is to extract as much useful information as possible from the message.

merbanan, 2018-12-03:
This is pretty much implemented. We have the mic field.

closed 2018-12-03 by merbanan