merbanan/rtl_433 #417: autoreconf fails - "possibly undefined macro: AC_PROG_LIBTOOL"

bigpresh, 2016-08-31:
Following the install instructions fails:

```
[davidp@supernova:~/rtl_433]$ autoreconf --install
configure.ac:16: error: possibly undefined macro: AC_PROG_LIBTOOL
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1
```

My checkout is at 0fb4dec498

Following is some version information in case it helps.  Am I missing something obvious?

```
[davidp@supernova:~/rtl_433]$ autoreconf --version
autoreconf (GNU Autoconf) 2.69
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+/Autoconf: GNU GPL version 3 or later
<http://gnu.org/licenses/gpl.html>, <http://gnu.org/licenses/exceptions.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by David J. MacKenzie and Akim Demaille.
[davidp@supernova:~/rtl_433]$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 8.5 (jessie)
Release:	8.5
Codename:	jessie
[davidp@supernova:~/rtl_433]$ aptitude show rtl-sdr | egrep '^(State|Version)'
State: installed
Version: 0.5.3-3
```

bigpresh, 2016-09-01:
Ah - I did not have libtool installed; installing that first made that unclear error go away.

Now, instead, I get as far as being able to run `./configure`, and getting:

```
./configure: line 11822: syntax error near unexpected token `LIBRTLSDR,'
./configure: line 11822: `PKG_CHECK_MODULES(LIBRTLSDR, librtlsdr)'
```

Installing librtlsdr-dev and re-running (from autoreconf onwards) made no apparent difference.

```
[davidp@supernova:~/rtl_433]$ aptitude search librtlsdr
i   librtlsdr-dev                                                                 - Software defined radio receiver for Realtek RTL2832U (development files)                
i A librtlsdr0                                                                    - Software defined radio receiver for Realtek RTL2832U (library)
```

I'm probably missing something obvious here.

panarom, 2016-09-14:
@bigpresh : I think I ran into this error at one point, and I believe this was the solution I found:
apt-get install pkg-config
make sure autoreconf -i is run (again) after installing package
HTH

bigpresh, 2016-09-15:
Ah - yes, after installing pkg-config, autoreconf --install works much better - still emits warnings, but they look harmless:

I was then able to run ./configure, which was happy, but `make` then bailed because libusb-1.0-0-dev wasn't installed - for anyone Googling the error, I got:

```
...
  CCLD     rtl_433
/usr/bin/ld: cannot find -lusb-1.0
collect2: error: ld returned 1 exit status
Makefile:602: recipe for target 'rtl_433' failed
make[2]: *** [rtl_433] Error 1
...
```

I installed `libusb-1.0-0-dev` (via apt-get) and was then able to compile and run rtl_433 - excellent stuff.

Thanks for the assistance, @panarom.

I think it would be nice if the build process was a bit more user-friendly and robust at detecting and reporting missing dependencies, etc (or, at least, if the dependencies were better documented in the install instructions) personally.  That's coming as a relatively experienced *nix user and developer, not a clueless newbie, too.

panarom, 2016-09-21:
The user-friendliest thing to do would probably be to package this up for the different distros. If I get really ambitious, I'll try to create a PPA one of these days.
@bigpresh : if you're satisfied with the resolution, can you close the issue? or @merbanan , if @bigpresh can't close it, I think you can; either way, I think it can be marked resolved.

bigpresh, 2016-09-22:
Sure thing, I'll close it, as it's working well for me now.  If I get a moment, I'd be happy to try to write up some more detailed install instructions along with a list of required packages etc.

closed 2016-09-22 by bigpresh

KimYar, 2021-11-03:
Hi everyone,
I am getting the below error msg after running buildozer -v android debug and I couldn't find a solution. Could you please help me with this?

`
`kivy is already unpacked, skipping
[INFO]:    <- directory context /Users/appleapple/Desktop/CalendarIA/CAI/.buildozer/android/platform/python-for-android
[INFO]:    # Prebuilding recipes
[INFO]:    Prebuilding hostpython3 for armeabi-v7a
[INFO]:    hostpython3 has no prebuild_armeabi_v7a, skipping
[INFO]:    Applying patches for hostpython3[armeabi-v7a]
[INFO]:    hostpython3 already patched, skipping
[INFO]:    Prebuilding libffi for armeabi-v7a
[INFO]:    libffi has no prebuild_armeabi_v7a, skipping
[INFO]:    Applying patches for libffi[armeabi-v7a]
[INFO]:    libffi already patched, skipping
[INFO]:    Prebuilding openssl for armeabi-v7a
[INFO]:    openssl has no prebuild_armeabi_v7a, skipping
[INFO]:    Prebuilding sdl2_image for armeabi-v7a
[INFO]:    sdl2_image has no prebuild_armeabi_v7a, skipping
[INFO]:    Applying patches for sdl2_image[armeabi-v7a]
[INFO]:    sdl2_image already patched, skipping
[INFO]:    Prebuilding sdl2_mixer for armeabi-v7a
[INFO]:    sdl2_mixer has no prebuild_armeabi_v7a, skipping
[INFO]:    Applying patches for sdl2_mixer[armeabi-v7a]
[INFO]:    sdl2_mixer already patched, skipping
[INFO]:    Prebuilding sdl2_ttf for armeabi-v7a
[INFO]:    sdl2_ttf has no prebuild_armeabi_v7a, skipping
[INFO]:    Prebuilding sqlite3 for armeabi-v7a
[INFO]:    sqlite3 has no prebuild_armeabi_v7a, skipping
[INFO]:    Prebuilding python3 for armeabi-v7a
[INFO]:    python3 has no prebuild_armeabi_v7a, skipping
[INFO]:    Applying patches for python3[armeabi-v7a]
[INFO]:    python3 already patched, skipping
[INFO]:    Prebuilding sdl2 for armeabi-v7a
[INFO]:    sdl2 has no prebuild_armeabi_v7a, skipping
[INFO]:    Prebuilding setuptools for armeabi-v7a
[INFO]:    setuptools has no prebuild_armeabi_v7a, skipping
[INFO]:    Prebuilding six for armeabi-v7a
[INFO]:    six has no prebuild_armeabi_v7a, skipping
[INFO]:    Prebuilding pyjnius for armeabi-v7a
[INFO]:    pyjnius has no prebuild_armeabi_v7a, skipping
[INFO]:    Applying patches for pyjnius[armeabi-v7a]
[INFO]:    pyjnius already patched, skipping
[INFO]:    Prebuilding android for armeabi-v7a
[INFO]:    android has no prebuild_armeabi_v7a, skipping
[INFO]:    -> directory context /Users/appleapple/Desktop/CalendarIA/CAI/.buildozer/android/platform/build-armeabi-v7a/build/other_builds/android-sdl2/armeabi-v7a__ndk_target_21/android
[INFO]:    <- directory context /Users/appleapple/Desktop/CalendarIA/CAI/.buildozer/android/platform/python-for-android
[INFO]:    Prebuilding kivy for armeabi-v7a
[INFO]:    kivy has no prebuild_armeabi_v7a, skipping
[INFO]:    # Building recipes
[INFO]:    Building hostpython3 for armeabi-v7a
[INFO]:    hostpython3 said it is already built, skipping
[INFO]:    Building libffi for armeabi-v7a
[INFO]:    libffi said it is already built, skipping
[INFO]:    -> running cp /Users/appleapple/Desktop/CalendarIA/CAI/.buildozer/android/platform/build...(and 219 more)
[INFO]:    Building openssl for armeabi-v7a
[INFO]:    openssl said it is already built, skipping
[INFO]:    -> running cp /Users/appleapple/Desktop/CalendarIA/CAI/.buildozer/android/platform/build...(and 391 more)
[INFO]:    Building sdl2_image for armeabi-v7a
[INFO]:    Building sdl2_mixer for armeabi-v7a
[INFO]:    Building sdl2_ttf for armeabi-v7a
[INFO]:    Building sqlite3 for armeabi-v7a
[INFO]:    sqlite3 said it is already built, skipping
[INFO]:    Building python3 for armeabi-v7a
[INFO]:    python3 said it is already built, skipping
[INFO]:    Building sdl2 for armeabi-v7a
[INFO]:    -> directory context /Users/appleapple/Desktop/CalendarIA/CAI/.buildozer/android/platform/build-armeabi-v7a/build/bootstrap_builds/sdl2/jni
[INFO]:    -> running ndk-build V=1 NDK_DEBUG=1
[INFO]:    <- directory context /Users/appleapple/Desktop/CalendarIA/CAI/.buildozer/android/platform/python-for-android       
[INFO]:    Building setuptools for armeabi-v7a
[... 292 lines cut ...]
#     XPC_FLAGS = '0x0'
#     PS1 = '(venv) \\h:\\W \\u\\$ '
#     _CE_M = ''
#     XPC_SERVICE_NAME = '0'
#     SHLVL = '1'
#     HOME = '/Users/appleapple'
#     CONDA_PYTHON_EXE = '/Users/appleapple/anaconda3/bin/python'
#     LOGNAME = 'appleapple'
#     LC_CTYPE = 'en_CA.UTF-8'
#     DISPLAY = '/private/tmp/com.apple.launchd.JG7odCqoLs/org.macosforge.xquartz:0'
#     _ = '/Users/appleapple/Desktop/CalendarIA/venv/bin/buildozer'
#     OLDPWD = '/Users/appleapple/Desktop/CalendarIA'
#     __PYVENV_LAUNCHER__ = '/Users/appleapple/Desktop/CalendarIA/venv/bin/python'
#     PACKAGES_PATH = '/Users/appleapple/.buildozer/android/packages'
#     ANDROIDSDK = '/Users/appleapple/.buildozer/android/platform/android-sdk'
#     ANDROIDNDK = '/Users/appleapple/.buildozer/android/platform/android-ndk-r19c'
#     ANDROIDAPI = '27'
#     ANDROIDMINAPI = '21'
# 
# Buildozer failed to execute the last command
# The error might be hidden in the log above this error
# Please read the full log, and search for it before
# raising an issue with buildozer itself.
# In case of a bug report, please add a full log with log_level = 2
``