merbanan/rtl_433 #419: Selectively disable devices that don't output structured data

rct, 2016-08-31:
When using structured data output (such as JSON, CSV, KV), there should be an option to disable devices that only output unstructured data in order to make it easier to parse.  This is for users who do not use -R to selectively enable only the desired devices.

One way to implement this would be to add a field to the device structure, that indicates whether structured data is output via data_make(). That flag would then be available as data for option parsing. 

Other flags/device classes might be desirable such as identifying devices that have protocols that don't allow good discrimination (via preamble, CRC, etc.) and yield false positives.

merbanan, 2016-09-09:
All devices that doesn't use structured data output are now disabled by default. There is a flag that can enable them all.

closed 2016-09-09 by merbanan

ludwich66, 2017-01-03:
Please see my entry in the google list
https://groups.google.com/forum/#!topic/rtl_433/yogNIYaHZOM
Happy New Year 
ludwich