merbanan/rtl_433 #424: higher samplerate -> no decoding (WT450)

dagbdagb, 2016-09-18:
With default samplerate, my WT450 decodes perfectly. In order to simultaneously pick up another sensor on a different frequency, I want to increase the samplerate.

But then the original signal is not decoded. See below for two runs detecting the same sensor, once with the default samplerate and once at 1 Msps.

**Default samplerate:**
rtl_433 -p 65 -f 433800000 -A
[....]
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000.
Bit detection level set to 8000.
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433800000 Hz.
Detected OOK package
2016-09-17 13:48:59 :   WT450 sensor
        House Code:      4
        Channel:         4
        Battery:         OK
        Temperature:     19.72 C
        Humidity:        81 %
Analyzing pulses...
Total count:   27,  width: 17725                (70.9 ms)
Pulse width distribution:
 [ 0] count:   16,  width:   239 [238;241]      ( 956 us)
 [ 1] count:   10,  width:   484 [484;485]      (1936 us)
 [ 2] count:    1,  width:   149 [149;149]      ( 596 us)
Gap width distribution:
 [ 0] count:   16,  width:   248 [247;250]      ( 992 us)
 [ 1] count:   10,  width:   492 [492;493]      (1968 us)
Pulse period distribution:
 [ 0] count:   10,  width:   487 [487;490]      (1948 us)
 [ 1] count:    4,  width:   977 [977;978]      (3908 us)
 [ 2] count:   12,  width:   732 [731;733]      (2928 us)
Level estimates [high, low]:  15819,   1335
Frequency offsets [F1, F2]:   -6244,      0     (-23.8 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with startbit/delimiter
Attempting demodulation... short_limit: 194, long_limit: 361, reset_limit: 494, demod_arg: 0
pulse_demod_pwm_ternary(): Analyzer Device 
bitbuffer:: Number of rows: 2 
[00] {26} 34 15 52 80 : 00110100 00010101 01010010 10
[01] {0} : 

**1M samplerate:**
rtl_433 -p 65 -f 433800000 -A -s 1000000
[....]
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000.
Bit detection level set to 8000.
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433800000 Hz.
[....]

Detected OOK package
Analyzing pulses...
Total count:   27,  width: 70895                (70.9 ms)
Pulse width distribution:
 [ 0] count:   16,  width:   951 [947;958]      ( 951 us)
 [ 1] count:   10,  width:  1930 [1928;1936]    (1930 us)
 [ 2] count:    1,  width:   585 [585;585]      ( 585 us)
Gap width distribution:
 [ 0] count:   16,  width:  1000 [996;1004]     (1000 us)
 [ 1] count:   10,  width:  1976 [1974;1981]    (1976 us)
Pulse period distribution:
 [ 0] count:   10,  width:  1952 [1948;1957]    (1952 us)
 [ 1] count:    4,  width:  3908 [3905;3910]    (3908 us)
 [ 2] count:   12,  width:  2929 [2927;2935]    (2929 us)
Level estimates [high, low]:  15887,   3538
Frequency offsets [F1, F2]:   -2433,      0     (-37.1 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with startbit/delimiter
Attempting demodulation... short_limit: 768, long_limit: 1440, reset_limit: 1982, demod_arg: 0
pulse_demod_pwm_ternary(): Analyzer Device 
bitbuffer:: Number of rows: 2 
[00] {26} 34 15 52 80 : 00110100 00010101 01010010 10
[01] {0} : 

Note the missing output right after 'Detected OOK package' in the last run.
Why is decoding unsuccessful when the samplerate is 1M?

zuckschwerdt, 2016-09-18:
the WT450 code is one of the few devices that uses PWM_Precise_Parameters, i.e pulse_tolerance (by way of OOK_PULSE_CLOCK_BITS, see pulse_demod_clock_bits()). The tolerance value is given in samples not in µs and thus locked to the sample rate. You can modify the code if you really need a different sample rate: https://github.com/merbanan/rtl_433/blob/master/src/devices/wt450.c#L138

merbanan, 2016-09-18:
@zuckschwerdt thanks for the info. Then then solution is to switch to µs and scale appropriate when needed.

merbanan, 2016-09-18:
@dagbdagb can you test by adding *4 to the values ?

dagbdagb, 2016-09-18:
@zuckschwerdt and @merbanan  Wonderful! 
Changing that one '20' into '80' enables me to receive multiple WT450s and my Efergy from a single rtl_433 instance running at 1Msps. 

The WT450s always transmit in triplicate, but now (at 1Msps) it looks like I only get one of the messages. 
Was I supposed to change these values as well to improve the ability to recognize the transmission?
```
   .short_limit   = 980,
   .long_limit    = 1952,
   .reset_limit   = 18000,
```
The output shown in my initial post appears to indicate that these values are calculated dynamically?
But the calculated values are not like the values in the code at either :

250ksps
`Attempting demodulation... short_limit: 194, long_limit: 361, reset_limit: 494, demod_arg: 0`

or

1Msps
`Attempting demodulation... short_limit: 768, long_limit: 1440, reset_limit: 1982, demod_arg: 0`

Thank you again @zuckschwerdt !

dagbdagb, 2016-09-18:
I realize that the thing with receiving only one of the datagrams may be because of two WT450s talking at almost the exact same time. Will check.

dagbdagb, 2016-09-19:
Getting all three datagrams now. I am happy. @merbanan : do you want to keep this issue open? Sounds like you want a change in the code?

merbanan, 2016-11-18:
@dagbdagb can you send some signal samples? The test repository doesn't have any.

merbanan, 2018-12-04:
@zuckschwerdt did you change the tolerances to us? Can this issue be closed?

zuckschwerdt, 2018-12-05:
Yes, this bug was fixed.

closed 2018-12-05 by zuckschwerdt