merbanan/rtl_433 #429: Option to execute a script on each signal

bigpresh, 2016-09-23:
I think it would be most useful for rtl_433 to have an argument to provide a script which would be executed each time a signal is received - so whenever a signal is received it would be executed and passed the data on STDIN.

I can, of course, rig up something myself, running rtl_433 with -F JSON, and each time a line is received, process it - but it would seem handy also to be able to just tell rtl_433 "whenever you receive something, pass it to this script".

Thoughts?

zuckschwerdt, 2016-09-23:
That would require job control on rtl_433's part (do we kill the script after some time, should we run multiple scripts in parallel, what to do about signals, ...)
And there is the gotcha that a single received packet could produce multiple events -- think weather station with multiple sensors.
Spawning on each event is the least favourable way to handle output. How about a read-loop in your script?

merbanan, 2016-09-23:
@bigpresh I will most likely not accept that kind of patch. A read loop script would do the same. I am more interested in a proper output module that can send events out to some kind of bus. mqtt, ubus, dbus etc. That way listeners can just hook into the message stream in a proper way.

bigpresh, 2016-09-23:
@zuckschwerdt :
> That would require job control on rtl_433's part (do we kill the script after some time, should we run multiple scripts in parallel, what to do about signals, ...)

I'd have thought just fork off and let the script be responsible for locking/job control/whatever.

> And there is the gotcha that a single received packet could produce multiple events -- think weather station with multiple sensors.

True.

> Spawning on each event is the least favourable way to handle output. How about a read-loop in your script?

Yeah, that's what I'm doing; was just thinking of another way that would work without having to have the script running all the time, and not requiring restarting the rtl_433 | script pipe each time I make a change.

Just an idle thought.

@merbanan:
> I am more interested in a proper output module that can send events out to some kind of bus. mqtt, ubus, dbus etc. That way listeners can just hook into the message stream in a proper way.

Yeah, agreed - firing to some bus/queue system would be nicer and more robust.

bigpresh, 2016-09-23:
By the way, as an aside, rtl_433 is working beautifully for me with an RTL2838 receiving signals from a THGR122N I picked up (after earlier attempts with a random cheap Chinese thing marked only "XT200" amounted to nothing).  Thanks for such an awesomely useful project.

closed 2016-09-23 by bigpresh