merbanan/rtl_433 #435: Request for support to implement SQL connectivity

tosko78, 2016-10-25:
hi, thanks for this useful software. I now would like to integrate temperature data to an MySQL database to pilot heating system and ensure most efficient energy use.

I started to add the required libraries and have an sql request ready to be used in print_csv_data function but facing issue to open SQL connection from there. I'm facing linkage issue and any MySQL related function fails within this function (although ok when I test SQL commands in the main section which confirms linkage looks ok)

I'm no longer good developpe I'm affraid and wonder if anybody could have a look at to either help me sorting out or could add this sql function to the tool ?

thanks in advance

panarom, 2016-12-08:
I'm pretty sure MySQL supports json now (I know Postgres does): can you write a shell script to pipe the json-output from rtl_433 into an insertion into mysql?
or switch to postgres/another json-capable database?

tosko78, 2017-01-10:
Yes it does , thanks and sorted now

Rgds

Sent from my iPhone ...

merbanan, 2018-12-04:
Issue solved. Closing.

closed 2018-12-04 by merbanan