merbanan/rtl_433 #445: Help with KKMoon PIR Detecctor

dejarikbcn, 2016-11-13:
Hello everyone, this is my first post.

I have purschased the following cheap PIR detector at Amazon. It works at 433MHZ:

[image]

I'm able to detect the signals sent from the sensor using **rtl_433 -a**. Those signals always follow this pattern:

One first row with a single 0:

**[00] {1} 00 : 0**

23 rows with always the same value:

**[01] {25} 09 c7 69 00 : 00001001 11000111 01101001 0
[02] {25} 09 c7 69 00 : 00001001 11000111 01101001 0
[03] {25} 09 c7 69 00 : 00001001 11000111 01101001 0
[04] {25} 09 c7 69 00 : 00001001 11000111 01101001 0
...
[23] {25} 09 c7 69 00 : 00001001 11000111 01101001 0**

And one last row with:

**[24] {89} 09 c7 69 04 e3 b4 82 71 da 41 38 80**

The rest of information from the signal is:

**signal_start = 382867, signal_end = 729882
signal_len = 347015,  pulses = 673
Iteration 1. t: 196    min: 100 (382)    max: 293 (291)    delta 9629
Iteration 2. t: 196    min: 100 (382)    max: 293 (291)    delta 0
Pulse coding: Short pulse length 100 - Long pulse length 293
Short distance: 87, long distance: 280, packet distance: 2964
p_limit: 196
bitbuffer:: Number of rows: 25**

I'm not really interested on decoding the signal. As this is just a simple PIR sensor: when the signal is detected, means that presence has been triggered. 

So I have implemented a simple **pir.c** program, added it to **/devices** folder, modified **rtl_433_devices.h**, **rtl_433.h**, etc... increased the MAX_PROTOCOLS variable, etc...

The program works fine. The device is identified.. However, I don't know how to use the program's callback method..

I JUST WANT to identify the sequence from the signal:

**09 c7 69 00 : 00001001 11000111 01101001 0**

But I don't know what I need to define the properties"modulation", "sort_limit", "long_limit", "reset_limit", or how to iterate through **bitbuffer** to detect that value:

**static int pir_callback(bitbuffer_t *bitbuffer) {
    //WHAT I NEED TO DO HERE??
    return 1;
}
r_device pir = {
    .name           = "PIR Sensor",
    .modulation     = OOK_PULSE_PWM_RAW ,
    .short_limit    = 100,
    .long_limit     = 300,
    .reset_limit    = 15000,
    .json_callback  = &pir_callback,
    .disabled       = 1,
    .demod_arg      = 0,
};**

Can anybody help me?

Thanks in advance!

zuckschwerdt, 2016-11-14:
You might want to code something like the Silvercrest device: https://github.com/merbanan/rtl_433/blob/master/src/devices/silvercrest.c -- it's pretty minimal.

dejarikbcn, 2016-11-14:
Thank you Christian.

I have had a look to that device. However, I don't know which value do i have to set to this properties:

_modulation 
short_limit 
long_limit 
reset_limit_ 

In order to get the same results than rtl_443 -a. Does anybody know which are te values that the flag analyzer mode uses?

By the way, I will try tonight with the same values than silvercrest:

**r_device silvercrest = {
    .name           = "Silvercrest Remote Control",
    .modulation     = OOK_PULSE_PWM_RAW,
    .short_limit    = 600,
    .long_limit     = 5000,
    .reset_limit    = 15000,
    .json_callback  = &silvercrest_callback,
    .disabled       = 1,
    .demod_arg      = 1,	// Remove startbit
};**

merbanan, 2016-11-14:
Run with -A it will give you a guess of the modulation used.

dejarikbcn, 2016-11-14:
Thank you Benjamin.

I will try to analyze with that flag tonight. I'll tell you the results.

dejarikbcn, 2016-11-14:
Hi Benjamin,

That's the output

```
Detected OOK package
Analyzing pulses...
Total count:   25,  width:  9295                (37.2 ms)
Pulse width distribution:
 [ 0] count:   14,  width:   102 [100;104]      ( 408 us)
 [ 1] count:   11,  width:   294 [292;296]      (1176 us)
Gap width distribution:
 [ 0] count:   13,  width:   280 [280;282]      (1120 us)
 [ 1] count:   11,  width:    89 [87;91]        ( 356 us)
Pulse period distribution:
 [ 0] count:   24,  width:   383 [381;384]      (1532 us)
Level estimates [high, low]:  16087,   3104
Frequency offsets [F1, F2]:   -6220,      0     (-23.7 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed period
Attempting demodulation... short_limit: 198, long_limit: 283, reset_limit: 283, demod_arg: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {25} f6 38 96 80 : 11110110 00111000 10010110 1
```

I don't understand. Running **rtl_433 -a** gives me:

`{25} 09 c7 69 00 : 00001001 11000111 01101001 0`

And **rtl_433 -A** gives me exactly the opposite bits:

`{25} f6 38 96 80 : 11110110 00111000 10010110 1`

Well, having this information analyzed, which are the best values to assing in my code?

```
.modulation
.short_limit
.long_limit
.reset_limit
```
Thank you!

merbanan, 2016-11-15:
Guessing modulation: Pulse Width Modulation with fixed period
OOK_PULSE_PWM_RAW
short_limit = (408 + 1176)/2 / 4
reset_limit = short_limit * 2

try that.

dejarikbcn, 2016-11-15:
hi benjamin,

you gave me values for short & reset limits. what about long limit?

and short limit you mean: (408+1176)/ 8?

thank you!!!

merbanan, 2016-11-15:
short_limit = (408+1176)/ 8 and set reset_limit  the same as long limit.

dejarikbcn, 2016-11-15:
Sorry no luck.

I'm repeating again the **rtl_433 -A** test, and triggering the PIR sensor. 

The package received is always the same as I mentioned before:

```
Detected OOK package
Analyzing pulses...
Total count:   25,  width:  9290                (37.2 ms)
Pulse width distribution:
 [ 0] count:   14,  width:   100 [97;104]       ( 400 us)
 [ 1] count:   11,  width:   290 [289;293]      (1160 us)
Gap width distribution:
 [ 0] count:   13,  width:   282 [281;286]      (1128 us)
 [ 1] count:   11,  width:    92 [90;94]        ( 368 us)
Pulse period distribution:
 [ 0] count:   24,  width:   383 [381;386]      (1532 us)
Level estimates [high, low]:  15943,   2682
Frequency offsets [F1, F2]:   -9963,      0     (-38.0 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed period
Attempting demodulation... short_limit: 195, long_limit: 287, reset_limit: 287, demod_arg: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {25} f6 38 96 80 : 11110110 00111000 10010110 1
```
I mean:

**[00] {25} f6 38 96 80 : 11110110 00111000 10010110 1**

But my code doesn't work. I'm just trying to print the buffer:

```
#include "rtl_433.h"

static int pir_callback(bitbuffer_t *bitbuffer) {
    bitbuffer_print(bitbuffer);
    return 1;
}

r_device pir = {
    .name           = "PIR Sensor",
    .modulation     = OOK_PULSE_PWM_RAW ,
    .short_limit    = 198,
    .long_limit     = 396,
    .reset_limit    = 396,
    .json_callback  = &pir_callback,
    .disabled       = 1,
    .demod_arg      = 0,
};
```
But when I trigger the motion, the **bitbuffer_print** method only prints buffers of 1 row with a single 0, repeatedly:

```
bitbuffer:: Number of rows: 1
[00] {1} 00 : 0
```

It is possibly due to my missknowledge of C. If you are agree, can we discuss it by email? I don't want to create too much useless posts here in your issue site.

Thank you!!!!

merbanan, 2016-11-15:
Try multiplying the values with 4.

merbanan, 2016-11-15:
And feel free to use the mailing list if you prefer email.

merbanan, 2016-11-15:
short_limit: 195, long_limit: 287, reset_limit: 287 multiplied by 4 will work.

dejarikbcn, 2016-11-15:
Multiplying by 4 worked like a charm!

The package was successfully identified:

```
bitbuffer:: Number of rows: 1
[00] {25} f6 38 96 80 : 11110110 00111000 10010110 1
```

With the values:

```
#include "rtl_433.h"

static int pir_callback(bitbuffer_t *bitbuffer) {
    bitbuffer_print(bitbuffer);
    return 1;
}

r_device pir = {
    .name           = "KKMoon PIR Sensor",
    .modulation     = OOK_PULSE_PWM_RAW ,
    .short_limit    = 792,
    .long_limit     = 1584,
    .reset_limit    = 1584,
    .json_callback  = &pir_callback,
    .disabled       = 1,
    .demod_arg      = 0,
};
```

So I'm able now to match the received buffer from what I'm expecting.

Thank you very much! I will send my C code when finished, if you want to include it in your devices list. (This KKMoon PIR sensor is usually in sale in Amazon/Ebay, and it's interesting to control the detected motion)

closed 2016-11-15 by dejarikbcn

merbanan, 2016-11-15:
This sensor should be a clone of  generic_remote.c. Just add the device declaration in that file and call the same callback and you should be good to go.

TonyHoyle, 2016-12-08:
As a data point I also have this device and the settings above work.  Mine broadcasts on 433.75 instead of 433.92, which caused some headscratching (they're cheap chinese junk really, but for home projects are really handy).

My raw data is slightly different, if it helps:

`Detected OOK package
Analyzing pulses...
Total count:   24,  width:  7791                (31.2 ms)
Pulse width distribution:
 [ 0] count:   10,  width:   244 [243;251]      ( 976 us)
 [ 1] count:   14,  width:    80 [79;83]        ( 320 us)
Gap width distribution:
 [ 0] count:    9,  width:    84 [82;87]        ( 336 us)
 [ 1] count:   14,  width:   246 [245;249]      ( 984 us)
Pulse period distribution:
 [ 0] count:   23,  width:   328 [325;333]      (1312 us)
Level estimates [high, low]:  15998,    103
Frequency offsets [F1, F2]:   -7160,      0     (-27.3 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed period
Attempting demodulation... short_limit: 162, long_limit: 250, reset_limit: 250, demod_arg: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {24} 2e c6 f6 : 00101110 11000110 11110110
`