merbanan/rtl_433 #452: Need some help to decode InFactory Temp/Hum Sensor (datas in rtl_443_test)

gbraux, 2016-12-03:
Hi !

I need some help/advices to decode a temperature/humidity 443 sensor. It is a "InFactroy TX05K-THC"
I have posted all details in the rtl_443_test repository, with pictures, datas, first findings (pull may not have been apprved yet, but you can get everything here : https://github.com/gbraux/rtl_433_tests/tree/master/tests/InFactroy-Temp-Humidity-Sensor-T05K-THC).

The current state :

Sensor is transmiting 5 bytes with PWM OOK encoding + a few 0000 for preamb. Those 5 bytes are transmitted 6 times, every minute. The PWM values looks the same as Esperanza EWS and Oregon, so RTL_433 tries to decode the frame using those decoders, but without success.

IIIIIIII ???????? ???????? ???????? ?????NNN

I = Sensor ID. Resets to a new value after battery is removed. N = Those 3 bits never ever changed

And some datas : 

```
<At home. Temp around 23C, Humidity around 30%. I may have blowed on the sensor at some time to have some value change>

[01] {40} 2d 70 6b 92 91 : 00101101 01110000 01101011 10010010 10010001
[01] {40} 2d c0 6b b2 81 : 00101101 11000000 01101011 10110010 10000001
[01] {40} 2d 61 6d 13 71 : 00101101 01100001 01101101 00010011 01110001
[01] {40} 2d 71 6d 84 41 : 00101101 01110001 01101101 10000100 01000001
[01] {40} 2d d2 6d 43 41 : 00101101 11010010 01101101 01000011 01000001
[01] {40} 2d 82 6c c3 01 : 00101101 10000010 01101100 11000011 00000001
[01] {40} 2d 22 6c c2 91 : 00101101 00100010 01101100 11000010 10010001

<Battery removed>

[01] {40} ab 60 6c a3 11 : 10101011 01100000 01101100 10100011 00010001
[01] {40} ab e0 6c c2 71 : 10101011 11100000 01101100 11000010 01110001
[01] {40} ab f0 6c 72 71 : 10101011 11110000 01101100 01110010 01110001
[01] {40} ab f2 6b 82 61 : 10101011 11110010 01101011 10000010 01100001
[01] {40} ab e2 6b 32 61 : 10101011 11100010 01101011 00110010 01100001

<Battery removed>

[01] {40} d0 a0 6b 32 71 : 11010000 10100000 01101011 00110010 01110001
[01] {40} d0 10 6b 12 61 : 11010000 00010000 01101011 00010010 01100001
[01] {40} d0 20 6b 12 51 : 11010000 00100000 01101011 00010010 01010001

<Battery removed + Placed the sensor in a very warm & wet environnement - T > 60C, H > 80%)

[00] {40} 5d 42 68 42 71 : 01011101 01000010 01101000 01000010 01110001
[00] {40} 5d c2 68 02 81 : 01011101 11000010 01101000 00000010 10000001
[00] {40} 5d b2 67 e3 51 : 01011101 10110010 01100111 11100011 01010001
[01] {40} 5d 71 81 a7 31 : 01011101 01110001 10000001 10100111 00110001
[00] {40} 5d 12 87 b5 81 : 01011101 00010010 10000111 10110101 10000001
[01] {40} 5d 72 7f a7 11 : 01011101 01110010 01111111 10100111 00010001
[01] {40} 5d d2 74 a7 81 : 01011101 11010010 01110100 10100111 10000001
[00] {40} 5d 92 66 f5 71 : 01011101 10010010 01100110 11110101 01110001
[00] {40} 5d 92 66 f5 71 : 01011101 10010010 01100110 11110101 01110001
[00] {40} 5d e2 66 a5 21 : 01011101 11100010 01100110 10100101 00100001
[00] {40} 5d d2 66 a5 11 : 01011101 11010010 01100110 10100101 00010001
[00] {40} 5d 62 66 a4 91 : 01011101 01100010 01100110 10100100 10010001
[00] {40} 5d 72 66 a4 81 : 01011101 01110010 01100110 10100100 10000001

<Battery removed>

[00] {40} 91 e0 68 43 41 : 10010001 11100000 01101000 01000011 01000001
[00] {40} 91 a0 68 53 51 : 10010001 10100000 01101000 01010011 01010001
[00] {40} 91 80 68 03 51 : 10010001 10000000 01101000 00000011 01010001
```

gbraux, 2016-12-04:
I think I got the humidity value ...

IIIIIIII ???????? ???????? ?????HHH HHHHNNNN

I = Sensor ID. Resets to a new value after battery is removed.
H = Humidiy on 7 bits (it went up to 115% when I grabbed values in a very wet envt, but I think those low cost hum sensors are only relevant on the 20-80% range ...)
N = Those 4 bits never ever changed

panarom, 2016-12-08:
Is there a channel select switch on the device? perhaps 1 2 or 3?
Out of idle curiosity, how/where were you testing this @ 60℃?
There might also be a checksum set of bits.
also, the humidity may be Binary Coded Decimal, so 2 4-bit sets correspond directly to a tens place decimal digit & a units place decimal digit; rather than 7 or 8 bits corresponding to a percentage (leading to percentages higher than 100)
I just went through this process with the Oregon Scientific SL109, and this looks similar; but unfortunately not similar enough to decode it directly.  I'll try and take a deeper look at your data in the near future. what's the feasibility of testing it in a refrigerator?  If you can gather data as it warms up or cools down to ambient temp, that will help.

ygator, 2016-12-10:
A couple more things to note.  The bits may need to be flipped and the temperature may have an offset added to it(lacrossews uses 30℃).

gbraux, 2016-12-13:
Panarom, Ygator, thx for you feedbacks !

FYI, I identified that there is effectively -25 offset for the humidity (125 means 100% humidity).
No progress on the temperature side, I am not an expert, so no coding technique is jumping out at me !
To grab high temperature values, I just put the sensor above boiling water (to simultate high humidity and high temperature).

As requested, I put the sensor in my freezer, and grabbed the datas bellow : 

`
------ 20 degrees Celcius to Fridge (10C) ------------
15 minutes

[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e1 64 53 51 : 01001011 11100001 01100100 01010011 01010001
[00] {40} 4b e2 63 e3 51 : 01001011 11100010 01100011 11100011 01010001
[00] {40} 4b e2 63 e3 51 : 01001011 11100010 01100011 11100011 01010001
[00] {40} 4b e2 63 e3 51 : 01001011 11100010 01100011 11100011 01010001
[00] {40} 4b e2 63 e3 51 : 01001011 11100010 01100011 11100011 01010001
[00] {40} 4b e2 63 e3 51 : 01001011 11100010 01100011 11100011 01010001
[00] {40} 4b e2 63 e3 51 : 01001011 11100010 01100011 11100011 01010001
[00] {40} 4b e2 63 e3 51 : 01001011 11100010 01100011 11100011 01010001
[00] {40} 4b e2 63 e3 51 : 01001011 11100010 01100011 11100011 01010001
[00] {40} 4b e2 63 e3 51 : 01001011 11100010 01100011 11100011 01010001
[00] {40} 4b e2 63 e3 51 : 01001011 11100010 01100011 11100011 01010001
[00] {40} 4b e2 63 e3 51 : 01001011 11100010 01100011 11100011 01010001
[00] {40} 4b e2 63 e3 51 : 01001011 11100010 01100011 11100011 01010001
[00] {40} 4b 92 62 93 51 : 01001011 10010010 01100010 10010011 01010001
[00] {40} 4b 92 62 93 51 : 01001011 10010010 01100010 10010011 01010001
[00] {40} 4b 92 62 93 51 : 01001011 10010010 01100010 10010011 01010001
[00] {40} 4b 92 62 93 51 : 01001011 10010010 01100010 10010011 01010001
[00] {40} 4b 92 62 93 51 : 01001011 10010010 01100010 10010011 01010001
[00] {40} 4b 92 62 93 51 : 01001011 10010010 01100010 10010011 01010001
[00] {40} 4b 92 62 93 51 : 01001011 10010010 01100010 10010011 01010001
[00] {40} 4b 92 62 93 51 : 01001011 10010010 01100010 10010011 01010001
[00] {40} 4b 92 62 93 51 : 01001011 10010010 01100010 10010011 01010001
[00] {40} 4b 92 62 93 51 : 01001011 10010010 01100010 10010011 01010001
[00] {40} 4b 92 62 93 51 : 01001011 10010010 01100010 10010011 01010001
[00] {40} 4b 92 62 93 51 : 01001011 10010010 01100010 10010011 01010001
[01] {40} 4b 02 61 03 51 : 01001011 00000010 01100001 00000011 01010001
[01] {40} 4b 02 61 03 51 : 01001011 00000010 01100001 00000011 01010001
[01] {40} 4b 02 61 03 51 : 01001011 00000010 01100001 00000011 01010001
[01] {40} 4b 02 61 03 51 : 01001011 00000010 01100001 00000011 01010001
[01] {40} 4b 02 61 03 51 : 01001011 00000010 01100001 00000011 01010001
[01] {40} 4b 02 61 03 51 : 01001011 00000010 01100001 00000011 01010001
[01] {40} 4b 02 61 03 51 : 01001011 00000010 01100001 00000011 01010001
[01] {40} 4b 02 61 03 51 : 01001011 00000010 01100001 00000011 01010001
[01] {40} 4b 02 61 03 51 : 01001011 00000010 01100001 00000011 01010001
[01] {40} 4b 02 61 03 51 : 01001011 00000010 01100001 00000011 01010001
[... 585 lines cut ...]
[01] {40} 4b e1 56 45 11 : 01001011 11100001 01010110 01000101 00010001
[01] {40} 4b e1 57 75 11 : 01001011 11100001 01010111 01110101 00010001
[01] {40} 4b e1 57 75 11 : 01001011 11100001 01010111 01110101 00010001
[01] {40} 4b e1 57 75 11 : 01001011 11100001 01010111 01110101 00010001
[01] {40} 4b e1 57 75 11 : 01001011 11100001 01010111 01110101 00010001
[01] {40} 4b e1 57 75 11 : 01001011 11100001 01010111 01110101 00010001
[01] {40} 4b e1 57 75 11 : 01001011 11100001 01010111 01110101 00010001
[01] {40} 4b e1 57 75 11 : 01001011 11100001 01010111 01110101 00010001
[01] {40} 4b e1 57 75 11 : 01001011 11100001 01010111 01110101 00010001
[01] {40} 4b e1 57 75 11 : 01001011 11100001 01010111 01110101 00010001
[... cut ...]
[01] {40} 4b e1 58 55 11 : 01001011 11100001 01011000 01010101 00010001
[01] {40} 4b e1 58 55 11 : 01001011 11100001 01011000 01010101 00010001
[01] {40} 4b e1 58 55 11 : 01001011 11100001 01011000 01010101 00010001
[01] {40} 4b e1 58 55 11 : 01001011 11100001 01011000 01010101 00010001
[01] {40} 4b e1 58 55 11 : 01001011 11100001 01011000 01010101 00010001
[01] {40} 4b e1 58 55 11 : 01001011 11100001 01011000 01010101 00010001
[01] {40} 4b e1 58 55 11 : 01001011 11100001 01011000 01010101 00010001`

panarom, 2016-12-17:
It seems like the middle byte correlates directly with the temperature; but I haven't figured out exactly how to calculate the temperature based off the byte's numeric value.  @gbraux do you have something like a kitchen thermometer, with which you could provide more accurate measurements of the ambient/refrigerator/freezer temperatures?  it looks like there's some uncertainty in the estimates above.

ygator, 2016-12-17:
Nibbles
01 23 45 67 89
-----------------
4b d2 5b 93 91
4b 22 5a 94 01

Nibble 7-8 appear to be BCD and is possibly your humidity.  As you can see above it rolled from 39 to 40 instead of 3A.

You say nibble 0-1 changes with power on so it appears to be the power on id.
Humidity nibble 7-8.
You say nibble 9 is always 1.
That leaves nibble 2-6 all or some of which would indicate the temperature.

gbraux, 2016-12-18:
Hi, and thanks for helping on this !

Just grabbed new data, and the exact temperature was 22.0 degrees Celsius. Humidity was under 20% (my other thermometer / humidity sensor do not show detailed humidity if under 20%).

[00] {40} 4b 70 64 63 51 : 01001011 01110000 01100100 01100011 01010001

panarom, 2016-12-19:
[00] {40} 4b 92 40 83 91 : 01001011 10010010 01000000 10000011 10010001
[00] {40} 4b 22 3f b3 91 : 01001011 00100010 00111111 10110011 10010001

3rd byte switches from a row of 0s to a row of 1s (at least for the last 6 bits), as the thermometer goes below 0 celsius, which seems like it has to be the transition point; but I still haven't figured out how the temperature is encoded. the 2nd bit also seems to be a 0 when the thermometer is below freezing, and a 1 otherwise; but that seems like a weird way to do it.

ygator, 2016-12-19:
@gbraux Are you reading the temperature and humidity from something that listens to this sensor or are you measuring it with something else?

rct, 2016-12-19:
@panarom 

> 3rd byte switches from a row of 0s to a row of 1s (at least for the last 6 bits), as the thermometer goes below 0 celsius, which seems like it has to be the transition point; but I still haven't [...]

I haven't really been looking at this, but what you are describing would make a lot more sense if the bits were inverted - one or two's complement arithmetic with a sign bit.

panarom, 2016-12-20:
@rct I considered that possibility, but I can't seem to get inverted values to match up any better than un-inverted ones to the supplied estimates of the actual temperature.
@gbraux : is there any chance you can ascertain a more precise estimate of the freezer temperature, and then leave the thermometer in there an hour or so before taking a reading from it?

gbraux, 2016-12-20:
@panarom,

Here are some new datas. I put the sensor in the fridge, then in the freeze, and left it about 1h before getting datas. I measured the exact temperature using my multimeter temp prope, so the value should be very precise (far more than 433Mhz sensor itself !).

T = +7.0C
[01] {40} 4b f2 53 d4 41 : 01001011 11110010 01010011 11010100 01000001

T = +6.9C
[01] {40} 4b 22 53 b4 41 : 01001011 00100010 01010011 10110100 01000001

T = -17.1C
[01] {40} 4b 92 37 34 41 : 01001011 10010010 00110111 00110100 01000001

T = -17.4C
[01] {40} 4b 82 36 b4 41 : 01001011 10000010 00110110 10110100 01000001

T= -17.7C
[01] {40} 4b 12 36 74 41 : 01001011 00010010 00110110 01110100 01000001

panarom, 2017-01-04:
ok, somebody check me on this (using @ygator 's designation): the formula for temperature is to concatenate nibbles 456 into 12-bit integer that counts the increments of .05℃, which I'll call T.
Then the temperature is 0.05 * T - 60.0 ℃

gbraux, 2017-01-05:
Hi @panarom,

Are you sure this is the right post ? ;-)
Your findings does not seems to be aligned with my mesurements (or I may be totaly dumb ...)

T = +7.0C (measured temp)
[01] {40} 4b f2 53 d4 41 : 01001011 11110010 01010011 11010100 01000001

Nibbles 4-5-6 : 0010-0101-0011
Decimal : 595
0.05 * 595 - 60 = -30.25 ℃ ...

Guillaume.

panarom, 2017-01-05:
You've got 3-4-5 (left most nibble is index 0, not 1), you should have  0101 0011 1101
Decimal: 1341
0.05 * 1341 - 60 = 7.05

panarom, 2017-01-05:
I suspect nibble 2 is a checksum, but I haven't yet worked out how it's calculated

panarom, 2017-09-25:
@gbraux somebody added an infactory decoder recently: https://github.com/merbanan/rtl_433/pull/592 does that work with your device?

merbanan, 2018-12-04:
rtl_433  -F csv -X 'n=name,m=OOK_PPM,s=1856,l=4072,r=8000,g=1800,t=250' will demodulate reference sample files.

merbanan, 2018-12-04:
And -R 91 will decode the samples.

merbanan, 2018-12-04:
I will fetch the test samples from the fork and see if I can enable the decoder by default.

merbanan, 2018-12-05:
Added preamble and more checks so that the inFactory decoder can be enabled by default.

merbanan, 2018-12-05:
CRC is still not understood. Closing issue.

closed 2018-12-05 by merbanan