merbanan/rtl_433 #454: hondaremote.c broken code

ms-jdow, 2016-12-07:
This line may not produce the desired results:
	device_id = ( bytes[ 44 ] >> 8 | bytes[ 45 ]);

A uint8_t shifted 8 bits to the right is always 0. Was "<<" meant instead of ">>" given that device_id is a uint16_t?

{^_^}

merbanan, 2016-12-07:
Most likely, there should be some hondaremote smaple available in the test repo. Try those and see if it works.

ms-jdow, 2016-12-08:
May I suggest somebody who is concerned about the Honda Remote do it? I decided I wanted a decoder for the Acurite 518346, which it doesn't seem to handle. Since I am working on Winders (hey, I make money this way) so I converted it over to Visual Studio and C++. This entailed "some editing", mostly explicit type casting and allocating/freeing huge buffers. Along the way one of the error reports suggested the bytes[44] >> 8 was a silly construct, at least in C++. So I thought I'd politely raise the issue. It should be a quick fix for somebody who is familiar with the code.

If I get decoding working on the 518346 I'll fold it back into this project or send it to somebody who can easier than I can.

{^_^} Just trying to be helpful.

{^_^}

merbanan, 2016-12-08:
Ok, I'll leave the ticket open for reference.

ukshady, 2016-12-08:
Thanks for the heads up, I will try and find some time to look at this.

rct, 2016-12-09:
Joanne (@ms-jdow) - I'm not familiar with the Acurite 518346 weather station, but this page, https://www.acurite.com/kbase/01099_Weather_Station.html, says the outside temperature/humidity is the 592TXR "Tower" sensor.   rtl_433 definitely decode that sensor.  It's protocol 39, so you need to use `-R 39` or `-G`.  You should find some samples in the rtl_433_tests repo. 

Is the version that you've gotten to compile under MS VC++ decoding other devices correctly?

Alternatively, have you tried compiling under MinGW?,  Compiled binaries are posted here: http://cognito.me.uk/computers/rtl_433-windows-binary-32-bit/, that you could test again.  Looks like the last time he compiled it was 11/11/2016.

If it's not working for that sensor, please open a new issue and provide pulse analyzer `-A` output and/or IQ samples `-a -t`.   Thanks.

ms-jdow, 2016-12-09:
-39 does decode it. It didn't classify correctly on the first dongle I tried it on. I should have tried one of the others. The dongle I chose to use is an E4000 that had worked with a previous decoder I hacked up in SDRSharp when it was still more or less open source. I added some code tailored for E4000 tuners to setup offset tuning. Now it works if I give it a good enough antenna location. I added the obvious lines below. They are just before async reading is started. Similar lines should go into the sync reading path, as well.
===8<---
			r = rtlsdr_set_center_freq( dev, frequency[ frequency_current ]);
			if ( r < 0 )
				fprintf( stderr, "WARNING: Failed to set center freq.\n" );
			else
				fprintf( stderr, "Tuned to %u Hz.\n", rtlsdr_get_center_freq( dev ));

			/*	jdow = useful code for E4000 based tuners. */
			r = rtlsdr_get_tuner_type( dev );
			if (( r != RTLSDR_TUNER_R820T )
			&&	( r != RTLSDR_TUNER_R828D ))
			{
				r = rtlsdr_set_offset_tuning( dev, 1 );
			}
			else
			{
				r = rtlsdr_set_offset_tuning( dev, 0 );
			}

			r = rtlsdr_read_async( dev
								 , rtlsdr_callback
								 , (void *) demod
===8<---

Now I can see what happens when I hit it with an X10 garage door opener.

Thank you very much for the pointer. I had gotten into a digression of hunting out memory leaks on exit. Some modest code movement and an if clause on the "out:" declaration mitigated what I found for the clean exit leak I saw. (I also found a leak I'd added to my rtlsdr.dll implementation for Windows. It's nice if you have a lot of dongles. But it's not completely up to date up here for some changes I made today. Tomorrow it will probably be OK.)

{^_^}

rct, 2016-12-09:
@ms-jdow - can you give any more background on this?  What's the effect (or error) if offset tuning doesn't get set for E4000's?  

Maybe I'm misunderstanding and I'm not looking at the code right now, why aren't you just testing for the E4000?

merbanan, 2016-12-09:
E4000 has a center spike, destoys reception. With offset tuning you take the spectrum with an offset to avoid the center spike. PR welcome.

ms-jdow, 2016-12-09:
Got it in one merbanan.

I debated on testing for E4000.

Actually the test is probably not needed, I believe. (This will bring it out if there is a bug in the FC drivers. They are supposed to benignly report an error.) The R820x dongles seem to be the most common ones today. So I left it at that - the results of a convenient cut and paste. Testing for E4000 would be technically better.

{^_^}

merbanan, 2018-12-04:
This hondaremote issue has be fixed some time ago. Closing issue.

closed 2018-12-04 by merbanan