merbanan/rtl_433 #456: Does DSC module decode all DSC sensors? Would it decode flood sensor?

loganto, 2016-12-09:
So DSC is described as:

>  "Registering protocol [23] "DSC Security Contact"

^which is imho seems like it only applies to a single device but in a device file it's described as:

> "DSC - Digital Security Controls 433 Mhz Wireless Security Contacts:  doors, windows, smoke, CO2, water"

^and this sounds like something else. Like it supports all those different types of sensors. Does anyone know which one is correct?

rct, 2016-12-09:
If you mean the WS4985 water/flood sensor, yes, it decodes that sensor.   It's basically the normally open contact version of the WS4945. plus the water probe module. 

The DSC code currently decodes the 433 Mhz DSC contacts (sensors) that are compatible with the RF receivers/transceivers RFK5500, TRS5164, TR5132) for Power Series panels (1832, 1864, 1616, etc.).   

That sensor family includes door/window contacts, motion and glass break sensors, smoke/CO2 detectors, water/flood detectors.   I've tested most all of those.   I've never tested the keyfobs since I don't have access to any of those.

I think the same sensors are used by the Alexor and Impassa panels from DSC.   Also, I believe ADT uses DSC equipment in some of their installations.

_NOTE: The Power Series NEO panel uses an entirely different (and supposedly more sophisticated/secure) wireless protocol which this will not decode._

loganto, 2016-12-09:
@rct yep WS4985 is the one I got.

closed 2016-12-09 by loganto

rct, 2016-12-09:
@loganto - Yes, those are tested, I have a couple of them.  

Note: The DSC code is spitting out the raw status byte,  I decode later in a wrapper script.   I'll convert that to JSON and add some bit decoding.

The bit you are interested in 0x02 is Closed (dry) or open/wet if not set.   

0x40 is set for heartbeat messages.  It is not set when events happen.