merbanan/rtl_433 #461: Mebus decoder formatting

marcus-aa, 2017-01-03:
Excellent software, thanks.

The Mebus decoder provides a differently formatted output than many other. 

- No timestamp
- No indentation

Also, the humidity is negative (at least when temp is negative)

Sensor event:
protocol       = Mebus/433
address        = 3
channel        = 1
battery        = Ok
unkown1        = 0
unkown2        = 15
temperature    = -5.4°C
humidity       = -83%
c3 8f ca fa d0

merbanan, 2017-01-13:
Feel free to fix it. Or sending test signals if they are missing.

bogdanr, 2017-03-05:
I have a remote for a WS8211 LED strip which gets detected incorrectly as this.
[This remote](http://bogdan.nimblex.net/assets/img/ws2811_remote.jpg) always sends `1000101100010000` as a preamble.

merbanan, 2017-03-05:
Please send signal samples.

bogdanr, 2017-03-05:
Two samples of the same button press:

https://drive.google.com/file/d/0B6iKR3kHjEsYVUZRZXZIOWNQM1k/view?usp=sharing
https://drive.google.com/file/d/0B6iKR3kHjEsYYUR4MUpVUUlrZWc/view?usp=sharing

```
*** signal_start = 2693251, signal_end = 2790196
signal_len = 96945,  pulses = 136
Iteration 1. t: 267    min: 138 (76)    max: 397 (60)    delta 58
Iteration 2. t: 267    min: 138 (76)    max: 397 (60)    delta 0
Pulse coding: Short pulse length 138 - Long pulse length 397

Short distance: 139, long distance: 395, packet distance: 1274

p_limit: 267
bitbuffer:: Number of rows: 5 
[00] {1} 80 : 1
[01] {34} b1 03 45 ba 40 : 10110001 00000011 01000101 10111010 01
[02] {34} b1 03 45 ba 40 : 10110001 00000011 01000101 10111010 01
[03] {34} b1 03 45 ba 40 : 10110001 00000011 01000101 10111010 01
[04] {33} b1 03 45 ba 00 : 10110001 00000011 01000101 10111010 0
*** signal_start = 3387979, signal_end = 3484909
signal_len = 96930,  pulses = 136
Iteration 1. t: 267    min: 138 (76)    max: 397 (60)    delta 40
Iteration 2. t: 267    min: 138 (76)    max: 397 (60)    delta 0
Pulse coding: Short pulse length 138 - Long pulse length 397

Short distance: 139, long distance: 395, packet distance: 1274

p_limit: 267
bitbuffer:: Number of rows: 5 
[00] {1} 80 : 1
[01] {34} b1 03 45 ba 40 : 10110001 00000011 01000101 10111010 01
[02] {34} b1 03 45 ba 40 : 10110001 00000011 01000101 10111010 01
[03] {34} b1 03 45 ba 40 : 10110001 00000011 01000101 10111010 01
[04] {33} b1 03 45 ba 00 : 10110001 00000011 01000101 10111010 0
```

merbanan, 2018-12-03:
Current code does not detect this signal. We need samples for all buttons on this remote to make something of it. Closing issue.

closed 2018-12-03 by merbanan