merbanan/rtl_433 #465: Man page/Release

martijnvanbrummelen, 2017-01-13:
Would it be possible to create a man-page for rtl_433 and please rename release "push" to 0.1 or higher so it can be  packaged for Debian.
Thanks!

merbanan, 2017-01-13:
Sure, can you point me in the correct direction for debian man page requirements and I will look into it?

I will also tag up a proper release point when the debian requirements are committed.

martijnvanbrummelen, 2017-01-13:
I can also create it and send a pull request?

merbanan, 2017-01-13:
Works for me.

martijnvanbrummelen, 2017-01-13:
Do you have a emailadres/alias I can put in the files?
If you don't want it on github send a mail to mvb@debian.org

merbanan, 2017-01-13:
banan at ludd dot ltu dot se

merbanan, 2018-12-03:
Maybe time for a 18.12 version. Anyway issue solved, closing.

closed 2018-12-03 by merbanan