merbanan/rtl_433 #466: acurite5n1 rain accumulation issue with multiple sensors

zerog2k, 2017-01-13:
rain accumulation logic breaks when more than one sensor reporting, as they are not tracked separately. Current code might mistake this for rain counter wrap-around.

example here:
https://gist.githubusercontent.com/zerog2k/6340a71252015d1032314b7e99d0b6e9/raw/e4bde902d8813a66d6b9935bc579be739d3b0742/gistfile1.txt

rct, 2017-01-13:
My vote is to just report the raw value (0.01 or whatever to get the units right).  Then clients can do what ever they want.  In general I think rtl_433, should hand the RF  bits, parse/validate athe framing to know when valid messages are received and then get out of the way.

If you are going to keep state and report deltas, then it should be as accurate and handle the edge cases such as multiple stations being in range, wraparounds, sensor reset, etc.

merbanan, 2017-01-13:
Please remove the state tracking in this case. I agree with @rct.