merbanan/rtl_433 #468: rtl_433 and 433Utils

kevinfoerster, 2017-01-17:
Hi,

i try to control my kitchen hood (http://elica.com/DE-de/products/hoods/seashell) using a 433mhz sender from a raspberry pi, i therefor checked with rtl_433 for some more details how to do this.

the hood comes with a rf remote and has a couple of buttons to turn the lights on/off, and start the fan. When i tried turning the light on/off, I received the following:
```
* signal_start = 852911, signal_end = 913052
signal_len = 60141,  pulses = 94
Iteration 1. t: 136    min: 92 (54)    max: 181 (40)    delta 8
Iteration 2. t: 136    min: 92 (54)    max: 181 (40)    delta 0
Distance coding: Pulse length 136

Short distance: 86, long distance: 175, packet distance: 3963

p_limit: 136
bitbuffer:: Number of rows: 5 
[00] {17} 49 7a 80 : 01001001 01111010 1
[01] {18} a4 bd 40 : 10100100 10111101 01
[02] {18} a4 bd 40 : 10100100 10111101 01
[03] {18} a4 bd 40 : 10100100 10111101 01
[04] {18} a4 bd 40 : 10100100 10111101 01
```

not sure what to do with this information and can this data somehow be used to send this data using 433Utils codesend script instead of using the remote?

I am just getting started here, and have no clue what I am doing ;)

merbanan, 2017-01-17:
Run rtl_433 -A and post the output. -A has a better analyzer.

kevinfoerster, 2017-01-17:
here we go

```
Tuned to 433920000 Hz.
Detected OOK package
Analyzing pulses...
Total count:   18,  width:  4640		(18.6 ms)
Pulse width distribution:
 [ 0] count:   10,  width:    91 [89;94]	( 364 us)
 [ 1] count:    8,  width:   180 [178;182]	( 720 us)
Gap width distribution:
 [ 0] count:    8,  width:    87 [86;88]	( 348 us)
 [ 1] count:    9,  width:   176 [175;178]	( 704 us)
Pulse period distribution:
 [ 0] count:    6,  width:   178 [176;180]	( 712 us)
 [ 1] count:    6,  width:   357 [356;358]	(1428 us)
 [ 2] count:    5,  width:   267 [264;270]	(1068 us)
Level estimates [high, low]:  15957,     95
Frequency offsets [F1, F2]:    -163,      0	(-0.6 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_limit: 91, long_limit: 0, reset_limit: 179, demod_arg: 0
pulse_demod_manchester_zerobit(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] {24} 26 64 12 : 00100110 01100100 00010010 

Detected OOK package
Analyzing pulses...
Total count:   19,  width:  4909		(19.6 ms)
Pulse width distribution:
 [ 0] count:   11,  width:    91 [89;94]	( 364 us)
 [ 1] count:    8,  width:   181 [178;183]	( 724 us)
Gap width distribution:
 [ 0] count:   10,  width:   175 [175;177]	( 700 us)
 [ 1] count:    8,  width:    86 [86;87]	( 344 us)
Pulse period distribution:
 [ 0] count:    6,  width:   267 [266;270]	(1068 us)
 [ 1] count:    6,  width:   178 [176;181]	( 712 us)
 [ 2] count:    6,  width:   357 [355;359]	(1428 us)
Level estimates [high, low]:  15909,     90
Frequency offsets [F1, F2]:    1791,      0	(+6.8 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_limit: 91, long_limit: 0, reset_limit: 178, demod_arg: 0
pulse_demod_manchester_zerobit(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] {25} 13 32 09 00 : 00010011 00110010 00001001 0

Detected OOK package
Analyzing pulses...
Total count:   19,  width:  4910		(19.6 ms)
Pulse width distribution:
 [ 0] count:   11,  width:    91 [89;94]	( 364 us)
 [ 1] count:    8,  width:   180 [177;182]	( 720 us)
Gap width distribution:
 [ 0] count:   10,  width:   176 [175;178]	( 704 us)
 [ 1] count:    8,  width:    87 [87;88]	( 348 us)
Pulse period distribution:
 [ 0] count:    6,  width:   268 [265;272]	(1072 us)
 [ 1] count:    6,  width:   177 [176;180]	( 708 us)
 [ 2] count:    6,  width:   356 [355;359]	(1424 us)
Level estimates [high, low]:  15893,     81
Frequency offsets [F1, F2]:    2354,      0	(+9.0 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_limit: 91, long_limit: 0, reset_limit: 179, demod_arg: 0
pulse_demod_manchester_zerobit(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] {25} 13 32 09 00 : 00010011 00110010 00001001 0

Detected OOK package
Analyzing pulses...
Total count:   19,  width:  4910		(19.6 ms)
Pulse width distribution:
 [ 0] count:   11,  width:    91 [89;95]	( 364 us)
 [ 1] count:    8,  width:   180 [177;183]	( 720 us)
Gap width distribution:
 [ 0] count:   10,  width:   176 [175;177]	( 704 us)
 [ 1] count:    8,  width:    87 [87;88]	( 348 us)
Pulse period distribution:
 [ 0] count:    6,  width:   268 [267;272]	(1072 us)
 [ 1] count:    6,  width:   178 [176;181]	( 712 us)
 [ 2] count:    6,  width:   355 [354;358]	(1420 us)
Level estimates [high, low]:  15940,     80
Frequency offsets [F1, F2]:    1965,      0	(+7.5 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_limit: 91, long_limit: 0, reset_limit: 178, demod_arg: 0
pulse_demod_manchester_zerobit(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] {25} 13 32 09 00 : 00010011 00110010 00001001 0

Detected OOK package
Analyzing pulses...
Total count:   19,  width:  4908		(19.6 ms)
Pulse width distribution:
 [ 0] count:   11,  width:    91 [89;96]	( 364 us)
 [ 1] count:    8,  width:   180 [178;183]	( 720 us)
Gap width distribution:
 [ 0] count:   10,  width:   176 [175;178]	( 704 us)
 [ 1] count:    8,  width:    87 [87;87]	( 348 us)
Pulse period distribution:
 [ 0] count:    6,  width:   267 [264;271]	(1068 us)
 [ 1] count:    6,  width:   178 [176;181]	( 712 us)
 [ 2] count:    6,  width:   356 [355;359]	(1424 us)
Level estimates [high, low]:  15927,    119
Frequency offsets [F1, F2]:     615,      0	(+2.3 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_limit: 91, long_limit: 0, reset_limit: 179, demod_arg: 0
pulse_demod_manchester_zerobit(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] {25} 13 32 09 00 : 00010011 00110010 00001001 0
```

merbanan, 2017-01-17:
There we go. So you need to send the manchester coded signal with the bitstream "13 32 09 00" to replicate this signal.

The 0 needs to be 364 us long and 1 needs to be 720 bits long. Try to send this signal with your tool and look at the output from rtl_433. When the output matches the receiver should accepts the signal.

merbanan, 2018-12-03:
One can now compare signals in pulseview via the support script. Closing issue.

closed 2018-12-03 by merbanan

f-666, 2020-02-02:
Did you ever manage to get the hood working?

kevinfoerster, 2020-02-02:
Unfortunately not… it is basically the only lightsource at home which can’t be remotely controlled.

To be honest, I did not understand what @merbanan was talking about… and I could not find the time looking into this issue… I would be happy to assist and test stuff if it helps.

kevinfoerster, 2022-04-27:
nice, I will give a try later today… this stupid remote still bugs me every day

TimelessNL, 2024-01-04:
Sorry to bump this old issue, but I'm also trying to control my kitchen hood.
Which to be honest is from Whirlpool but as it is a Elica white-label product this issue seems to be the right place.

I'm using one of those cheap `WL102-341` transmitters and an implementation based on ESPHome.
when I compare the remote's MC encoded signal with my own they are both identical yet the device does not respond at all.

When I look at the rtl_433 output of the two I do see something different though:
Remote:
`Frequency offsets [F1, F2]:     494,      0     (+1.9 kHz, +0.0 kHz)`
My signal:
`Frequency offsets [F1, F2]:   -5368,      0     (-20.5 kHz, +0.0 kHz)`

I assume this is the specific frequency the devices are operating in the 433MHz band?
And based on these results they are 18kHz apart?
Could that be why it doesn't work?

zuckschwerdt, 2024-01-04:
> they are 18kHz apart? Could that be why it doesn't work?
No.

Don't use rtl_433 to compare signals. You'll only see the best-effort extracted data portion. You might get some insight with the raw .ook output. But compare the actual spectrum, e.g. https://triq.org/pdv/

magnayn, 2024-01-04:
Your hood might use different codes, but I figured out the packets to send by attaching one of those cheap 433Mhz receivers to a cheap logic analyzer (basically a salae clone), and then just outputting the equivalent by bit-banging an I/O pin attached to the equivalent transmitter unit. It wasn't super-sensitive to timings - though the actual original remote isn't amazingly reliable either!

TimelessNL, 2024-01-04:
The manual indeed mentions that the hood has to be power cycled in order to '_link_' with a remote, so the codes may differ because of that. But unless the communication is bidirectional (_which I doubt_) I suppose the uniqueness can only be set in the factory.

I will give both options a try, thanks for the suggestions!
Unfortunately the 433MHz band is very busy around here and distinguishing signals by hand is almost impossible but I can always give it a try.

I must say that I have a different experience, my remote is always spot on.
For the 2 years I've been using it it never missed a beat.
Don't know if it may help but this is the remote I'm working with:
[image]

magnayn, 2024-01-04:
Mine looks more like this: https://www.elica.com/IT-en/remote-control-0

433Mhz is noisy everywhere - we very occasionally find the fan / light has turned itself on, which I assume is it being accidentally parsing noise as a valid command.

I think in the case of my remote, the protocol follows an <address><command> format, so the 'pairing' process is probably the remote 'learning' which address is being broadcast by the hood.

That said, I have worked with other 433Mhz devices that I struggled with, only to realise the pairing process was about resetting the sequence in a 'cycling code' style remote -- I.E: the same button sends very different codes each time, either for security reasons, or simply to make it hard for 3rd parties to work with their products.

It might be worth just double-checking if you're able that your style of remote isn't doing code-cycling

TimelessNL, 2024-01-04:
Ah we do have different remotes indeed. I don't think mine uses any rolling code or code cycling mechanism as when I hold the button or press it repeatedly it always send out the same sequence for each function. I did notice that only the last 3 bits differ tough. For example Lights `ON` ends with `111` and light `OFF` is `000`

I also checked `triq.org`, which without me knowing I already did 😅 as `rtl_433` conveniently provides a clickable link during analyzing. And apart from some +/- 4µs timing differences between the signals they are identical.

TimelessNL, 2024-01-05:
Found the issue. 🥳 

I limited the rtl_sdr reception by removing its antenna. This gave me the possibility to better analyze the data transmitted be the remote as is was close by. And at that moment I noticed that all commands are repeated 4 times. So I altered my ESPHome configuration to send each command 4 times with a 15ms interval and this made it work every try (_based on limited testing_)

So yeah.. command was right all along, it only needed to be repeated. Which seems common in 433MHz land?

magnayn, 2024-01-06:
Because the receiver can't know how strong a signal represents '1', it has automatic gain control. It's common to need to 'bias' the the receiver, either by sending multiple times or by sending a pattern that's roughly 50% on and 50% off (e.g 10101010).

This is why most protocols tend to use something like Manchester encoding.