merbanan/rtl_433 #471: need a better mechansim for identifying units

matthewwall, 2017-01-18:
there is no way to know what units a decoder is sending.  the rtl_433 command has an option for units:

    -C (native | si | customary)

but what are the units of 'native'?  that is not clear from the decoder output, nor is it clear where one would go to figure it out.

what are the units of 'customary'?

also, does 'si' really mean 'si', as in meter instead of kilometer for every measure of length?  it would help both the writers of the decoders and the users of their output to know exactly which units will be returned, e.g., centimeter vs meter vs millimeter vs kilometer.  does 'si' mean that rain will be reported in millimeter but lightning strike distance will be reported in kilometers?

the decoder output should be self-identifying wrt units and unit system - the consumer of the decoder output might have no idea what was specified to the rtl_433 program.

one approach is to include a 'unit_system' in the json output.  this parameter would have a value of METRIC, METRICWX, or US.  each of these unit systems then has a clear definition of which units will be used for each type of measurement.

as things are now in rtl_433, ensuring consistent units is a nightmare.

zuckschwerdt, 2017-01-18:
Quick note:
- **native** is whatever the device delivers to the decoder
- **si** is a currently a conversion from °F to °C
- **customary** (you might know this as US, Imperial, …) is a currently a conversion from °C to °F
It would be nice to convert e.g. rain measurements, we'd need a way for decoders to report the unit first.
The JSON output will show the unit of temperature readings on the key as "_F" or "_C" postfix. This isn't perfect but other options like nested structures "{value: x, unit: y}" or unrelated keys "temperature:x, temperature_unit: y" are not much better.

matthewwall, 2017-01-18:
proposal:

start by enumerating the actual units to be used for 'si' and 'customary'.  do this in the 'how to write a decoder' guide for rtl_433 so that writers of the decoders (or those who want to fix them) know what to write.  the unit groups and their associated units would be:
```
temperature - C or F
depth (e.g., snow, rain) - mm or inch
distance (e.g., distance to lightning strike) - km or mile
altitude - meter or foot
pressure - mbar or inHg
humidity - percent
rain rate - mm/period or inch/period
```
then there are (at least) two options for communicating the units.

option 1: use the append-units-to-label pattern in every decoder.  for example:

    {"time": xxxx, "temperature_C": z}
    {"time": xxxx, "temperature_F": y} # never just 'temperature'
    {"time": xxxx, "rain_mm"}
    {"time": xxxx, "rain_inch"} # never just 'rain'

option 2: include unit system in the json packet.  for example:

    {"time": xxxx, "temperature": z, "pressure": y, "unit_system": "SI"}
    {"time": xxxx, "temperature": z, "pressure": y, "unit_system": "US"}

which assumes that SI and US are defined something like:

    "SI": {temperature: degree_C, pressure: mbar}
    "US": {temperature: degree_F, pressure: inHG}

option 1 is probably less hassle, more explicit, less overhead, and reasonably future-proof.

m

merbanan, 2017-01-18:
I will not approve option 1. The -C parameter is for that. If you use that all temperature readings will be in the form the user wants them. For other values this has not yet been implemented. And as far as I know the unit is appended in the temperature json data. Native units is what the transmitter sends.

merbanan, 2017-01-18:
If you are up to write a json schema describing all parameters that would be very helpful. The idea that the output should be self describing to what extent possible is good. If you start proposing things and send some patches I am sure we can come up with something that is good in all aspects.

matthewwall, 2017-01-19:
perhaps i did not explain option 1 very well.  it would still require the -C parameter.  for example, if you specified `-C si`, then you would get:

    {"time": xxxx, "temperature_C": z, "wind_speed_kph": y}

whereas if you specified `-C customary`, the same decoder would emit:

    {"time": xxxx, "temperature_F": r, "wind_speed_mph": s}

so the only things we need to establish are:

* what are the units for the si and customary unit systems, and
* what are the strings the decoders should use to represent those units

each decoder *should* document what the units are for each observation when 'native' is specified, but since the labels are self-describing, that documentation is less critical.  for example, `-C native` might result in:

    {"time": xxxx, "temperature_F": q, "wind_speed_mps": n}

if the native units are degree F for temperatures and meters-per-second for velocities.

happy to work on a json schema for the option 2 approach, but i thought this might be simpler while still solving the problem.

merbanan, 2018-12-04:
@zuckschwerdt how about dropping the native option? I sort of can't find a use for it, that way this is not an issue. And then default to si units. You have added the config file so it should be easy for US people to change the global default for their use.

zuckschwerdt, 2018-12-05:
I always used `-C si` (and `-q`) but I'd rather not change the coded default or disable the option to get the "raw" output whatever that is.

But The new config file option makes it very easy to have sane defaults now. Just use a `~/.rtl_433.conf` with:
```
convert si
```
The example config has that line already :)

As the output harmonisation (#827) is progressing well, I think this is solved.

closed 2018-12-05 by zuckschwerdt

zuckschwerdt, 2018-12-05:
quick note: current conversion includes:
- "F" - "C"
- "mph" - "kph"
- "inch" - "mm"
- "inHg" - "hPa"
- "PSI" - "kPa"