merbanan/rtl_433 #472: ELV WS2000 decoding does not work.

marcus-aa, 2017-01-19:
I have an older but quite common system (ELV WS2000/2500/3000) with a lot of sensors; S2000I, S2000IA, S2000W etc. They are all mentioned in the code elv.c but nothing is decoded.

I have tried both -G and -R 07.

-A produces nice OOK packages, but it is like it never calls the ELV decoder.

Detected OOK package
Analyzing pulses...
Total count:   61,  width: 18405		(73.6 ms)
Pulse width distribution:
 [ 0] count:   38,  width:   211 [210;215]	( 844 us)
 [ 1] count:   23,  width:    88 [88;91]	( 352 us)
Gap width distribution:
 [ 0] count:   38,  width:    93 [92;95]	( 372 us)
 [ 1] count:   22,  width:   216 [215;218]	( 864 us)
Pulse period distribution:
 [ 0] count:   60,  width:   305 [304;308]	(1220 us)
Level estimates [high, low]:  15879,      4
Frequency offsets [F1, F2]:    -184,      0	(-0.7 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed period
Attempting demodulation... short_limit: 149, long_limit: 219, reset_limit: 219, demod_arg: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1 
[00] {61} 00 31 e9 47 10 84 3b a8 

Next transmission: [00] {61} 00 31 e8 47 10 84 3f a8

Other protocols (Mebus, RT-tech etc) work fine, so the setup is working.

Any ideas?

/M

merbanan, 2017-01-19:
Submit some test signals and hope some kind soul will add the needed code.

rct, 2017-01-19:
Please do submit some samples to rtl_433_tests if there aren't ones there already.

Take a look at the timing limits (device structure) in the source code for that decoder.  Compare those to what the pulse analyzer (as shown above) is receiving.   Those timing limits decide whether the decoder is going to get called.

Given that it appears to be a fairly early protocol (number 7) it is possible/likely that the timing limits were set with a specific level limit (-l nnnn) very low or high that doesn't match what the current auto-level code sees.   IIRC vestom made some adjustments to timings when auto level was implemented if and only if, there were samples in the rtl_433_tests repo.   So make sure there are good/valid signal recordings in rtl_433_tests. 

If you want to try some experiments without changing the code, you could try setting a manual level limit (-l) to see if that triggers the decoder.  Try low levels (1000-6000) and high values (10000 - 15000)

marcus-aa, 2017-01-19:
> Submit some test signals and hope some kind soul will add the needed code.

Will do, the code is there so should hopefully only need some debugging. Will also try rct's suggestions, thanks.

Is there an easy way to add the test files without cloning the whole test repository? I'm running this on a headless Rpi and there is just not enough room for all the current files. Not so familiar with git...

p1mike, 2017-02-21:
Same problem for me. I created and attached some sample files or you could download here:
https://www.dropbox.com/s/pk0e2c6edvzwmre/samples.zip?dl=0
Pls tell me if you got the samples.
[samples.zip](https://github.com/merbanan/rtl_433/files/789635/samples.zip)

vestom, 2017-03-30:
Added the samples to https://github.com/merbanan/rtl_433_tests/pull/151
Made a small fix to the ELV driver in https://github.com/merbanan/rtl_433/pull/508
It looks like it could use some love from someone with the sensors though...
Issue can be closed

marcus-aa, 2017-05-02:
Thanks vestom, now the decoding works. I will try to verify the sensor values and add correct handling of the rain sensor values.

closed 2017-05-02 by marcus-aa