merbanan/rtl_433 #478: Cent-A-Meter power monitor

voltagex, 2017-01-28:
```
*** signal_start = 6505921, signal_end = 6558497
signal_len = 52576,  pulses = 116
Iteration 1. t: 191    min: 127 (98)    max: 255 (18)    delta 9217
Iteration 2. t: 191    min: 127 (98)    max: 255 (18)    delta 0
Pulse coding: Short pulse length 127 - Long pulse length 255

Short distance: 115, long distance: 116, packet distance: 238

p_limit: 191
bitbuffer:: Number of rows: 18 
[00] {24} 80 00 00 : 10000000 00000000 00000000 
[01] {1} 80 : 1
[02] {1} 80 : 1
[03] {1} 80 : 1
[04] {13} 00 10 : 00000000 00010
[05] {1} 80 : 1
[06] {4} 20 : 0010
[07] {4} 40 : 0100
[08] {2} 40 : 01
[09] {10} 00 40 : 00000000 01
[10] {3} 20 : 001
[11] {3} 40 : 010
[12] {5} 10 : 00010
[13] {2} 80 : 10
[14] {34} 00 00 00 08 00 : 00000000 00000000 00000000 00001000 00
[15] {5} 40 : 01000
[16] {2} 40 : 01
[17] {1} 80 : 1
*** signal_start = 20008282, signal_end = 20060860
signal_len = 52578,  pulses = 111
Iteration 1. t: 192    min: 129 (88)    max: 255 (23)    delta 9608
Iteration 2. t: 192    min: 129 (88)    max: 255 (23)    delta 0
Distance coding: Pulse length 192

Short distance: 114, long distance: 115, packet distance: 237

p_limit: 192
bitbuffer:: Number of rows: 23 
[00] {18} ff ff c0 : 11111111 11111111 11
[01] {0} : 
[02] {0} : 
[03] {0} : 
[04] {1} 80 : 1
[05] {9} ff 80 : 11111111 1
[06] {0} : 
[07] {2} c0 : 11
[08] {1} 80 : 1
[09] {0} : 
[10] {1} 80 : 1
[11] {6} fc : 111111
[12] {0} : 
[13] {3} e0 : 111
[14] {1} 80 : 1
[15] {1} 80 : 1
[16] {1} 80 : 1
[17] {25} ff ff ff 80 : 11111111 11111111 11111111 1
[18] {0} : 
[19] {1} 80 : 1
[20] {0} : 
[21] {0} : 
[22] {2} c0 : 11
```

Arduino code is at https://github.com/jonoxer/CentAReceiver

merbanan, 2017-01-28:
https://github.com/merbanan/rtl_433#supporting-additional-devices-and-test-data

voltagex, 2017-01-28:
Apologies, I'll close this for now - I'm not even sure the device is transmitting correctly.

closed 2017-01-28 by voltagex