merbanan/rtl_433 #487: pulse_FSK_detect(): Maximum number of pulses reached! Error

mnikser, 2017-02-24:
Hello,

Sorry if it is not the right place for this question.
I am using rtl_433 to collect temperature data from a few Ambient weather sensors.
I have started a get this error now and the program does not receive any data anymore.
pulse_FSK_detect(): Maximum number of pulses reached

I am trying to understand what causes this error.
I basically have no clue what it means.
Is it because of hardware issues or software problems.

Any help/directions is appreciated,
--
Mohammad

vestom, 2017-03-04:
The error means the program is apparently receiving a very long FSK encoded signal with more bits/pulses than any currently known devices. However the Ambient sensors are nor FSK encoded, but OOK encoded instead.
A likely cause is that your sensors are now sending their signal at the same time, disturbing the  transmission of each other. As the sensors send at slightly different frequencies it is received as something that might look like a FSK encoded signal. It will happen as the periodic transmission from the sensors drift slightly and becomes overlapping, and it will disappear again as the timing drift further.
Either that or a bug in the program :-)
However without any more information (e.g. a recording or analysis of the received signal) it is not possible to look further into it, and I recommend closing this issue.

mnikser, 2017-03-05:
Hi Vestom,

Thanks for your reply.
I am not sure that is the problem in my case.
I have 5 sensors and all five transmitting at the same time is odd. I actually reset one sensor and the same problem. It looks like my rtl receiver has a problem. It is a cheaper one from ebay.
I replaced it with a NooElec stick and it is working now.
I will see if the same thing will happen or not.

Thanks,
--
Mohammad

closed 2017-03-05 by mnikser