merbanan/rtl_433 #488: Accurite 6045M Lightning Detector Bad Temp

n7qnm, 2017-02-26:
Using an Acurite 6045M Lightning Detector it's supposed to provide Temp and Humidity - humidity looks OK but temp appears to be decoding incorrectly  - I have it in a room that is between 65 and 75F; RTL decodes as 35 to 258.   Would be happy to help debug with some assistance.

clayj@nwlink.com

rct, 2017-02-27:
I wasn't able to figure out how the temperature was encoded in my 6045M lightning detector and eventually gave up, suspecting that my sensor was bad or I'm missing something obvious. 

If you can provide output with some clear observations maybe we can get this resolved.

rct, 2017-03-19:
Fixed by #500.   This should be closed.

rct, 2018-04-22:
@zuckschwerdt @merbanan @n7qnm the temperature decoding was fixied 13 months ago. I believe you can close this issue.

closed 2018-04-22 by zuckschwerdt