merbanan/rtl_433 #510: Fix for battery status on Acurite 606TX

skiphansen, 2017-04-02:
Fixes the Acurite 606TX battery status which was being reported incorrectly as "Low" when it was in fact "Ok".

Thanks for the code!
Skip

`From 05fd64ef5f3da6eeba714b9ad348fc679c05efbf Mon Sep 17 00:00:00 2001
From: Skip Hansen <shansen@gfrn.org>
Date: Sun, 2 Apr 2017 14:16:00 -0700
Subject: [PATCH] Fix battery status on Acurite 606TX.

---
 src/devices/acurite.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/src/devices/acurite.c b/src/devices/acurite.c
index 506dd4c..75d88de 100755
--- a/src/devices/acurite.c
+++ b/src/devices/acurite.c
@@ -732,7 +732,7 @@ static int acurite_606_callback(bitbuffer_t *bitbuf) {
 
       	    temperature = temp / 10.0;
 	    sensor_id = bb[1][0];
-	    battery = bb[1][1] & 0x8f >> 7;
+	    battery = (bb[1][1] & 0x80) >> 7;
 
 	    data = data_make("time",          "",            DATA_STRING, time_str,
                              "model",         "",            DATA_STRING, "Acurite 606TX Sensor",
-- 
2.7.4
`

merbanan, 2017-04-02:
Send a PR and I will merge it.

skiphansen, 2017-04-03:
Sorry, the current HEAD didn't build using ./configure;make so I had pulled an earlier version.  My fix is already in the HEAD.  BTW: cmake does build the current HEAD.   I was on autopiliot and didn't notice cmake.

closed 2017-04-03 by skiphansen