merbanan/rtl_433 #512: LaCrosse TX141TH-Bv2 temperature calculation

craigthom, 2017-04-05:
It's not my sensor, but apparently one of my neighbors has one of these.  When run from the prompt rtl_433 shows this:

 2017-04-05 07:56:35
        Temperature in deg F:    138.56 F
        Humidity:        58 %
        Sensor ID:       da
:       LaCrosse TX141TH-Bv2 sensor
        Battery:         LOW
        Test?:   Yes

That humidity is a bit lower than my sensors show, but the temperature is crazy

I think the sensor is outputting F, but it is being treated as C and converted to F.

craigthom, 2017-04-05:
Now I'm picking up another of the same sensors with a different ID.  The temperature is also off, but not as much.

Since I don't have access to these sensors I may be picking up something else entirely.  There's just not enough data to go on.

closed 2017-04-05 by craigthom

zuckschwerdt, 2021-01-27:
This might have been something like a Bresser 3CH. The TX141TH-Bv2 now has a checksum validation to prevent false positives.