merbanan/rtl_433 #515: Signal too strong?

ichilton, 2017-04-12:
Hi,

I was using a DVB dongle with a wire ariel and was picking up all of my CurrentCost units except for one which was furthest away.

I replaced the dongle with one with an external antenna and now i'm picking up all of the units at a distance, but it's not picking up all of the ones in the same room.

If I take a transmitter out of the room and put it at the other side of the house, it starts picking it up.

Is the signal too strong and some how overloading it? - is there anything I can do?

I've tried using -g and putting in different values but it doesn't seem to make a difference.

Thanks,

Ian

strowger, 2017-05-15:
I see something like this too - if the currentcost tx is very nearby, it doesn't always read.

It's intermittent - for example, I have a pair of IAMs in here about a meter from the aerial on the RTL device, one of which reads all the time, the other hardly at all.

I'd assumed it was due to some sort of frequency tolerance issue but you're quite right that it could be a signal strength issue.

I worked around by using more than one RTL device/receiver; this issue is very low priority for me.

zuckschwerdt, 2017-05-16:
Try to capture the signal with `rtl_sdr` or `rtl_433` and look at the waveform in e.g. audacity.

merbanan, 2018-12-03:
There is SNR output in the analyzer (-A) now. Use that to compare signal strength. Closing issue.

closed 2018-12-03 by merbanan