merbanan/rtl_433 #516: Oregon Scientific WGR968 Wind Direction

PuceBaboon, 2017-04-15:
There appears to be a nibble-ordering issue with the wind direction (quadrant) code for the Oregon Scientific WRG968 anemometer.  With the current code, rtl_443 reports compass degree readings in excess of 900, with wild swings within a few seconds.  When checking against the degree indicator on the Oregon weather-station display, it quickly became apparent that the digits in the rtl_433 quadrant code were transposed, ie:-
 
                  rtl_433 shows direction as "902", but weather-station shows 290.
                  rtl_433 shows direction as "113", but weather-station shows 311.
                  rtl_433 shows direction as "213", but weather-station shows 321.
                  rtl_433 shows direction as "183", but weather-station shows 318.

At line 255 in devices/oregon_scientific.c the line:-

`float quadrant = (((msg[4] &0x0f)*100)+((msg[4]>>4)*10) + ((msg[5]>>4)&0x0f));`

...should be replaced with:-

`float quadrant = (((msg[4] &0x0f)*10) + ((msg[4]>>4)&0x0f) + (((msg[5]>>4)&0x0f) * 100));`

The test files demonstrating this are available under rtl_433_tests/tests/oregon_scientific/07.

merbanan, 2017-04-15:
Please send a PR.

merbanan, 2018-12-09:
This was merged sometime ago. Closing issue.

closed 2018-12-09 by merbanan