merbanan/rtl_433 #541: Pls add Auriol V2 protokol

directman66, 2017-05-19:
From RFLINK

Auriol V2:
Z31055A-TX, Z31055B-TX version 09/2012, Xiron, Watshome YT6018-2, Technoline,
AOK-2026, AOK-2011, Tunex MF-0211, Jula / Marquant 943134

example libruary for link http://www.meteocercal.info/forum/Thread-AuriolDuino-RX-Latest-Software

directman66, 2017-05-19:
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
Detected FSK package	@ 2017-05-19 09:03:45
Analyzing pulses...
Total count:  244,  width: 51953		(207.8 ms)
Pulse width distribution:
 [ 0] count:   19,  width:    11 [10;13]	(  44 us)
 [ 1] count:   43,  width:    53 [44;64]	( 212 us)
 [ 2] count:   41,  width:   168 [133;203]	( 672 us)
 [ 3] count:    8,  width:    15 [14;17]	(  60 us)
 [ 4] count:   15,  width:    22 [18;26]	(  88 us)
 [ 5] count:   13,  width:    29 [25;34]	( 116 us)
 [ 6] count:   13,  width:   315 [269;401]	(1260 us)
 [ 7] count:   15,  width:   238 [202;283]	( 952 us)
 [ 8] count:   20,  width:    75 [58;91]	( 300 us)
 [ 9] count:   35,  width:   106 [85;126]	( 424 us)
 [10] count:   18,  width:    39 [33;43]	( 156 us)
 [11] count:    1,  width:   754 [754;754]	(3016 us)
 [12] count:    3,  width:   442 [435;454]	(1768 us)
Gap width distribution:
 [ 0] count:   39,  width:    90 [74;112]	( 360 us)
 [ 1] count:   28,  width:   132 [113;161]	( 528 us)
 [ 2] count:   30,  width:    36 [29;44]	( 144 us)
 [ 3] count:   28,  width:   186 [151;226]	( 744 us)
 [ 4] count:   36,  width:    55 [45;66]	( 220 us)
 [ 5] count:   21,  width:    25 [21;28]	( 100 us)
 [ 6] count:    1,  width:   643 [643;643]	(2572 us)
 [ 7] count:    9,  width:    10 [10;12]	(  40 us)
 [ 8] count:   14,  width:   304 [260;369]	(1216 us)
 [ 9] count:   17,  width:    17 [13;20]	(  68 us)
 [10] count:    8,  width:   234 [219;245]	( 936 us)
 [11] count:    7,  width:    70 [68;74]	( 280 us)
 [12] count:    5,  width:   453 [416;491]	(1812 us)
Pulse period distribution:
 [ 0] count:   41,  width:    99 [79;118]	( 396 us)
 [ 1] count:   34,  width:   138 [124;172]	( 552 us)
 [ 2] count:   76,  width:   240 [184;299]	( 960 us)
 [ 3] count:    9,  width:    50 [44;55]	( 200 us)
 [ 4] count:    7,  width:   621 [552;713]	(2484 us)
 [ 5] count:   39,  width:   365 [311;459]	(1460 us)
 [ 6] count:    6,  width:    35 [32;38]	( 140 us)
 [ 7] count:   17,  width:    72 [65;77]	( 288 us)
 [ 8] count:    8,  width:   182 [175;189]	( 728 us)
 [ 9] count:    4,  width:   485 [470;505]	(1940 us)
 [10] count:    1,  width:   939 [939;939]	(3756 us)
 [11] count:    1,  width:    27 [27;27]	( 108 us)
Level estimates [high, low]:   3477,      1
Frequency offsets [F1, F2]:    7365,  -7086	(+28.1 kHz, -27.0 kHz)
Guessing modulation: No clue...

Detected FSK package	@ 2017-05-19 09:03:45
Analyzing pulses...
Total count:  434,  width: 87592		(350.4 ms)
Pulse width distribution:
 [ 0] count:   24,  width:    13 [11;16]	(  52 us)
 [ 1] count:   78,  width:    91 [73;112]	( 364 us)
 [ 2] count:    7,  width:    10 [10;10]	(  40 us)
 [ 3] count:   70,  width:    46 [35;56]	( 184 us)
 [ 4] count:   54,  width:    65 [55;77]	( 260 us)
 [ 5] count:   66,  width:   140 [116;169]	( 560 us)
[... 390 lines cut ...]
 [10] count:   22,  width:   206 [199;215]	( 824 us)
 [11] count:   29,  width:   405 [330;558]	(1620 us)
 [12] count:    4,  width:   627 [606;674]	(2508 us)
 [13] count:    5,  width:    52 [52;52]	( 208 us)
Pulse period distribution:
 [ 0] count:    3,  width:    19 [17;21]	(  76 us)
 [ 1] count:  296,  width:   224 [180;283]	( 896 us)
 [ 2] count:  226,  width:   107 [84;133]	( 428 us)
 [ 3] count:  184,  width:   338 [273;424]	(1352 us)
 [ 4] count:  119,  width:    65 [51;82]	( 260 us)
 [ 5] count:   16,  width:    26 [22;32]	( 104 us)
 [ 6] count:  201,  width:   155 [131;180]	( 620 us)
 [ 7] count:   62,  width:   514 [428;647]	(2056 us)
 [ 8] count:   11,  width:   698 [648;787]	(2792 us)
 [ 9] count:   38,  width:    44 [34;52]	( 176 us)
 [10] count:   31,  width:    82 [76;85]	( 328 us)
 [11] count:   11,  width:    33 [30;35]	( 132 us)
 [12] count:    1,  width:   942 [942;942]	(3768 us)
Level estimates [high, low]:   3665,     19
Frequency offsets [F1, F2]:    6997,  -7453	(+26.7 kHz, -28.4 kHz)
Guessing modulation: No clue...

Signal caught, exiting!

User cancel, exiting...

directman66, 2017-05-19:
Registered 57 out of 78 device decoding protocols
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
Detected FSK package	@ 2017-05-19 09:12:29
Analyzing pulses...
Total count:  359,  width: 71102		(284.4 ms)
Pulse width distribution:
 [ 0] count:    1,  width:     0 [ 0; 0]	(   0 us)
 [ 1] count:   20,  width:    10 [10;12]	(  40 us)
 [ 2] count:   37,  width:    32 [27;39]	( 128 us)
 [ 3] count:   45,  width:   193 [158;242]	( 772 us)
 [ 4] count:   56,  width:    62 [49;77]	( 248 us)
 [ 5] count:   33,  width:    22 [18;27]	(  88 us)
 [ 6] count:   17,  width:   266 [244;314]	(1064 us)
 [ 7] count:   56,  width:    94 [78;116]	( 376 us)
 [ 8] count:   39,  width:   135 [118;156]	( 540 us)
 [ 9] count:   10,  width:    13 [13;16]	(  52 us)
 [10] count:   10,  width:   432 [360;535]	(1728 us)
 [11] count:    6,  width:    17 [17;18]	(  68 us)
 [12] count:   23,  width:    44 [40;49]	( 176 us)
 [13] count:    5,  width:   342 [335;349]	(1368 us)
 [14] count:    1,  width:     4 [ 4; 4]	(  16 us)
Gap width distribution:
 [ 0] count:   53,  width:    30 [24;37]	( 120 us)
 [ 1] count:   66,  width:    97 [79;121]	( 388 us)
 [ 2] count:   29,  width:   253 [202;311]	(1012 us)
 [ 3] count:   12,  width:    15 [13;18]	(  60 us)
 [ 4] count:   61,  width:    64 [52;78]	( 256 us)
 [ 5] count:   17,  width:    11 [10;12]	(  44 us)
 [ 6] count:   22,  width:    21 [19;24]	(  84 us)
 [ 7] count:   45,  width:   168 [139;205]	( 672 us)
 [ 8] count:   16,  width:   127 [120;136]	( 508 us)
 [ 9] count:   29,  width:    44 [38;51]	( 176 us)
 [10] count:    5,  width:   345 [318;389]	(1380 us)
 [11] count:    3,  width:   508 [456;562]	(2032 us)
Pulse period distribution:
 [ 0] count:   17,  width:    43 [36;51]	( 172 us)
 [ 1] count:   77,  width:   118 [92;147]	( 472 us)
 [ 2] count:   26,  width:    60 [52;74]	( 240 us)
 [ 3] count:   69,  width:   288 [233;361]	(1152 us)
 [ 4] count:   19,  width:    79 [74;89]	( 316 us)
 [ 5] count:   94,  width:   178 [143;223]	( 712 us)
 [ 6] count:   32,  width:   446 [366;569]	(1784 us)
 [ 7] count:   15,  width:   227 [221;232]	( 908 us)
 [ 8] count:    6,  width:   610 [560;663]	(2440 us)
 [ 9] count:    3,  width:    26 [23;28]	( 104 us)
Level estimates [high, low]:   4578,      2
Frequency offsets [F1, F2]:    7401,  -7389	(+28.2 kHz, -28.2 kHz)
Guessing modulation: No clue...

Detected FSK package	@ 2017-05-19 09:12:30
Analyzing pulses...
Total count:  686,  width: 141895		(567.6 ms)
Pulse width distribution:
 [ 0] count:  137,  width:    61 [46;75]	( 244 us)
 [ 1] count:   76,  width:    30 [24;37]	( 120 us)
 [ 2] count:   50,  width:   243 [192;299]	( 972 us)
 [ 3] count:   24,  width:   343 [284;421]	(1372 us)
 [ 4] count:  106,  width:   134 [109;177]	( 536 us)
 [ 5] count:   44,  width:    42 [37;48]	( 168 us)
[... 3028 lines cut ...]
 [ 9] count:    2,  width:    19 [18;20]	(  76 us)
 [10] count:    1,  width:     1 [ 1; 1]	(   4 us)
Gap width distribution:
 [ 0] count:    1,  width:    19 [19;19]	(  76 us)
 [ 1] count:    2,  width:    11 [11;11]	(  44 us)
 [ 2] count:    4,  width:    26 [24;29]	( 104 us)
 [ 3] count:    3,  width:   138 [116;157]	( 552 us)
 [ 4] count:    1,  width:   205 [205;205]	( 820 us)
 [ 5] count:    3,  width:    84 [77;96]	( 336 us)
 [ 6] count:    3,  width:    61 [56;67]	( 244 us)
 [ 7] count:    2,  width:    45 [44;46]	( 180 us)
 [ 8] count:    1,  width:   107 [107;107]	( 428 us)
Pulse period distribution:
 [ 0] count:    1,  width:    19 [19;19]	(  76 us)
 [ 1] count:    4,  width:   216 [184;235]	( 864 us)
 [ 2] count:    2,  width:    77 [73;82]	( 308 us)
 [ 3] count:    4,  width:   123 [116;129]	( 492 us)
 [ 4] count:    3,  width:   361 [338;385]	(1444 us)
 [ 5] count:    3,  width:   165 [156;170]	( 660 us)
 [ 6] count:    1,  width:    36 [36;36]	( 144 us)
 [ 7] count:    1,  width:   272 [272;272]	(1088 us)
 [ 8] count:    1,  width:    45 [45;45]	( 180 us)
Level estimates [high, low]:   3352,    644
Frequency offsets [F1, F2]:    7043,  -7226	(+26.9 kHz, -27.6 kHz)
Guessing modulation: No clue...

zuckschwerdt, 2017-05-19:
The pulse analysis is not useful. The mentioned code is in a RAR which can only be downloaded by members and there is no licence note. If you want this to happen: get (-a -t or use rtl_sdr) a few (~3?) samples, check if they look clean (I use audacity) and commit them to rtl_433_tests with a description of the readings. Then maybe descibe the protocol here. Give us modulation, timing, information in the bits/bytes, crc at least.

directman66, 2017-05-19:
thanks
[2.zip](https://github.com/merbanan/rtl_433/files/1013250/2.zip)

zuckschwerdt, 2017-05-19:
Your signals are very noisy. Please open in Audacity (Import Raw, Unsigned 8-bit, 2 channels). The packet needs to be visible with significant amplitude. And when you zoom in the pulses should stand out. See the picture at http://triq.net/articles/sdr-nyquist-plugin-am-decode-iq-signals for an example.

directman66, 2017-05-19:
pi@raspberrypi:~ $ rtl_sdr -a -t
rtl_sdr: invalid option -- 'a'
rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:   -f frequency_to_tune_to [Hz]
        [-s samplerate (default: 2048000 Hz)]
        [-d device_index (default: 0)]
        [-g gain (default: 0 for auto)]
        [-p ppm_error (default: 0)]
        [-b output_block_size (default: 16 * 16384)]
        [-n number of samples to read (default: 0, infinite)]
        [-S force sync output (default: async)]
        filename (a '-' dumps samples to stdout)

directman66, 2017-05-19:
why command for need raw signal

zuckschwerdt, 2017-05-19:
Perhaps use `rtl_sdr -f 433920000 -s 1024000 -n 3072000 auriol-v2_1024k_$(date +%s).cu8` for a 3 second recording. Use without `-n <num>` for a open-end recording and stop with ctrl-C.
Please look at you signals in Audacity before you send them. Maybe cut out just the interesting part. I mostly record 10 minutes or so then pick a good part of myabe 0,5 seconds.

zuckschwerdt, 2017-05-19:
If you are doing this on a Raspi then `-s 1024000` might not work. Too much data. Just leave it out for default of 250000. (and use _250k_ in the filename!)

directman66, 2017-05-19:
why program you part raw signal?

directman66, 2017-05-19:
i see amplifier

directman66, 2017-05-19:
[image]

bad signal?

zuckschwerdt, 2017-05-19:
Select with the mouse. Then use File -> Export Selected (RAW, Unsigned 8 bit PCM).
Use the zoom button. It looks too weak. Should have much more amplitude.

zuckschwerdt, 2017-05-19:
Compare "Step 3. Analyze" here: http://mightydevices.com/?p=300
and "Analyzing the Signal" here: https://ilias.giechaskiel.com/posts/rtl_433/

directman66, 2017-05-19:
https://yadi.sk/mail/?hash=qFI7vilI1TOsBX00%2FHZK4qndBIc3tZ8OXnT1SWsEvP0%3D

directman66, 2017-05-19:
I'm not yet ready to implement it myself, but thanks very much for the links, very interesting

zuckschwerdt, 2017-05-19:
The signal is still very noisy and hard to see. Try to get the sender closer to the receiver and do not plug the rtl receiver directly into the Raspi, use a short USB cable to reduce noise from the Raspi.

There are 4 transmissions with the packet repeated identical 12 times per transmission.
Simple OOK pulse coding with short and long gaps, the sample has 37 pulses per packet.

Do you have the reading for this packet? Is it just degrees celsius or an additional hygrometer reading?

directman66, 2017-05-19:
https://yadi.sk/mail/?hash=9o069u%2FbYioJhzG%2FOT0e0S%2FcAYKvv2Dad2E0kY2Ku5Q%3D

directman66, 2017-05-19:
25,7 grad 39% hum

directman66, 2017-05-19:
[image]

zuckschwerdt, 2017-05-19:
Great. The signal is good now. I'll implement an experimental decoder in the next few days.

directman66, 2017-05-19:
BIG thanks friend!!!

zuckschwerdt, 2017-05-19:
Hold on. This is already handled by the Nexus decoder. The signal was just to weak before :)
```
rtl_433 -f 433920000 -s 250000 -q -F json -r auriol-v2_250k_good.cu8

{"time" : "@2.359296s", "model" : "Nexus Temperature/Humidity", "id" : 158, "battery" : "OK", "channel" : 2, "temperature_C" : 25.700, "humidity" : 40}
{"time" : "@5.242880s", "model" : "Nexus Temperature/Humidity", "id" : 158, "battery" : "OK", "channel" : 2, "temperature_C" : 25.700, "humidity" : 40}
{"time" : "@6.815744s", "model" : "Nexus Temperature/Humidity", "id" : 158, "battery" : "OK", "channel" : 2, "temperature_C" : 25.700, "humidity" : 39}
```
Temperature and Humidity match up.

zuckschwerdt, 2017-05-19:
Happy to help. Have fun!

directman66, 2017-05-19:
big thanks!!!!!!!!

zuckschwerdt, 2017-05-20:
@merbanan resolved and can be closed

closed 2017-05-20 by merbanan