merbanan/rtl_433 #543: Will there be support for switches 433 RCSwict ?

directman66, 2017-05-19:


zuckschwerdt, 2017-05-19:
If you are referring to the RC-Switch lib, I guess the protocol from e.g. https://github.com/sui77/rc-switch/blob/master/RCSwitch.cpp#L52 could be turned into receiver code.
But this will need a bunch of good signals or preferably someone with access to that hardware.

directman66, 2017-05-19:
Equipment is available in large quantities. As I will be at home - I will be able to lay out a dump.

directman66, 2017-05-20:
https://yadi.sk/mail/?hash=orDFAAtgvtnHivw0eHm%2B9QMDtK8lQd2ovad2H536Jbk%3D link to raw damp rc-switch

zuckschwerdt, 2017-05-20:
Interesting. There are 3 types of packets in there: one packet from Auriol/Nexus (ASK), one run of packets from the rc-switch (ASK), and then 7 very high frequency packets (FSK), like position beacons.

zuckschwerdt, 2017-05-20:
The RC_Switch encoding is straight forward. It's PCM transmission with 18 repeats (rows) of 25 bits packets.
The pulses are 1745 us wide with long pulses of 1347 us + 398 us gap, and short pulses of 440 us + 1305 us gap. The packet gap is 13239 us.
The transmission is repeated about 11 times, the first and last are truncated (less rows).
I'll try to give you a decoder.

zuckschwerdt, 2017-05-20:
A quick hack is here: https://github.com/zuckschwerdt/rtl_433/tree/feat-rcswitch (abusing the Silvercrest code for a quick test -- not for general use). I.e.
```
git clone -b feat-rcswitch https://github.com/zuckschwerdt/rtl_433.git rtl_433_rcswitch
cd rtl_433_rcswitch
mkdir build
cd build
cmake ..
make
```
Test some buttons, send me the output with your notes (Button 1 On: Code abc, Button 1 Off: Code def, ...)

directman66, 2017-05-20:
With which key should you run rtl_433 of this repository? While I get only:

pi@raspberrypi:~/rtl_433_rcswitch/build/src $ ./rtl_433
Registering protocol [1] "RC-Switch Remote Control"
Registering protocol [2] "Rubicson Temperature Sensor"
Registering protocol [3] "Prologue Temperature Sensor"
Registering protocol [4] "Waveman Switch Transmitter"
Registering protocol [5] "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol [6] "Acurite 609TXC Temperature and Humidity Sensor"
Registering protocol [7] "Oregon Scientific Weather Sensor"
Registering protocol [8] "Mebus 433"
Registering protocol [9] "KlikAanKlikUit Wireless Switch"
Registering protocol [10] "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol [11] "Fine Offset Electronics, WH2 Temperature/Humidity Sensor"
Registering protocol [12] "Nexus Temperature & Humidity Sensor"
Registering protocol [13] "Ambient Weather Temperature Sensor"
Registering protocol [14] "Calibeur RF-104 Sensor"
Registering protocol [15] "GT-WT-02 Sensor"
Registering protocol [16] "Danfoss CFR Thermostat"
Registering protocol [17] "Chuango Security Technology"
Registering protocol [18] "Generic Remote SC226x EV1527"
Registering protocol [19] "TFA-Twin-Plus-30.3049 and Ea2 BL999"
Registering protocol [20] "Fine Offset Electronics WH1080/WH3080 Weather Station"
Registering protocol [21] "WT450"
Registering protocol [22] "LaCrosse WS-2310 Weather Station"
Registering protocol [23] "Esperanza EWS"
Registering protocol [24] "Generic temperature sensor 1"
Registering protocol [25] "WG-PB12V1"
Registering protocol [26] "HIDEKI TS04 Temperature, Humidity, Wind and Rain Sensor"
Registering protocol [27] "Watchman Sonic / Apollo Ultrasonic / Beckett Rocket oil tank monitor"
Registering protocol [28] "CurrentCost Current Sensor"
Registering protocol [29] "emonTx OpenEnergyMonitor"
Registering protocol [30] "HT680 Remote control"
Registering protocol [31] "S3318P Temperature & Humidity Sensor"
Registering protocol [32] "Akhan 100F14 remote keyless entry"
Registering protocol [33] "Quhwa"
Registering protocol [34] "OSv1 Temperature Sensor"
Registering protocol [35] "Proove"
Registering protocol [36] "Bresser Thermo-/Hygro-Sensor 3CH"
Registering protocol [37] "Springfield Temperature and Soil Moisture"
Registering protocol [38] "Oregon Scientific SL109H Remote Thermal Hygro Sensor"
Registering protocol [39] "Acurite 606TX Temperature Sensor"
Registering protocol [40] "TFA pool temperature sensor"
Registering protocol [41] "Kedsum Temperature & Humidity Sensor"
Registering protocol [42] "blyss DC5-UK-WH (433.92 MHz)"
Registering protocol [43] "Steelmate TPMS"
Registering protocol [44] "Schrader TPMS"
Registering protocol [45] "Elro DB286A Doorbell"
Registering protocol [46] "Efergy Optical"
Registering protocol [47] "Honda Car Key"
Registering protocol [48] "Fine Offset Electronics, XC0400"
Registering protocol [49] "Radiohead ASK"
Registering protocol [50] "Kerui PIR Sensor"
Registering protocol [51] "Fine Offset WH1050 Weather Station"
Registering protocol [52] "Honeywell Door/Window Sensor"
Registering protocol [53] "Maverick ET-732/733 BBQ Sensor"
Registering protocol [54] "LaCrosse TX141TH-Bv2 sensor"
Registering protocol [55] "Acurite 00275rm,00276rm Temp/Humidity with optional probe"
Registering protocol [56] "LaCrosse TX35DTH-IT Temperature sensor"
Registering protocol [57] "LaCrosse TX29IT Temperature sensor"
Registering protocol [58] "Vaillant calorMatic 340f Central Heating Control"
Registering protocol [59] "Fine Offset Electronics, WH25 Temperature/Humidity/Pressure Sensor"
Registering protocol [60] "Fine Offset Electronics, WH0530 Temperature/Rain Sensor"
Registering protocol [61] "IBIS beacon"
Registered 61 out of 80 device decoding protocols
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
pulse_FSK_detect(): Maximum number of pulses reached!
pulse_FSK_detect(): Maximum number of pulses reached!
pulse_FSK_detect(): Maximum number of pulses reached!
pulse_FSK_detect(): Maximum number of pulses reached!
pulse_FSK_detect(): Maximum number of pulses reached!

zuckschwerdt, 2017-05-20:
What you tried should just work. For a quick test try
```
rtl_433 -s 1024000 -q -F json -r rc-switch_1024k_1495263163.cu8
```
Best to always use `-q -F json` for cleaner output.
Maybe use `-s 1024000` but that should not be necessary.

directman66, 2017-05-20:
login as: pi
pi@192.168.1.35's password:
Access denied
pi@192.168.1.35's password:
Access denied
pi@192.168.1.35's password:
Access denied
pi@192.168.1.35's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat May 20 17:35:01 2017
pi@raspberrypi:~ $
pi@raspberrypi:~ $ mc

pi@raspberrypi:~/rtl_433_rcswitch/build/src $ ./
CMakeFiles/ rtl_433
pi@raspberrypi:~/rtl_433_rcswitch/build/src $ ./rt
bash: ./rt: Нет такого файла или каталога
pi@raspberrypi:~/rtl_433_rcswitch/build/src $ ./rtl_433
Registering protocol [1] "RC-Switch Remote Control"
Registering protocol [2] "Rubicson Temperature Sensor"
Registering protocol [3] "Prologue Temperature Sensor"
Registering protocol [4] "Waveman Switch Transmitter"
Registering protocol [5] "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol [6] "Acurite 609TXC Temperature and Humidity Sensor"
Registering protocol [7] "Oregon Scientific Weather Sensor"
Registering protocol [8] "Mebus 433"
Registering protocol [9] "KlikAanKlikUit Wireless Switch"
Registering protocol [10] "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol [11] "Fine Offset Electronics, WH2 Temperature/Humidity Sensor"
Registering protocol [12] "Nexus Temperature & Humidity Sensor"
Registering protocol [13] "Ambient Weather Temperature Sensor"
Registering protocol [14] "Calibeur RF-104 Sensor"
Registering protocol [15] "GT-WT-02 Sensor"
Registering protocol [16] "Danfoss CFR Thermostat"
Registering protocol [17] "Chuango Security Technology"
Registering protocol [18] "Generic Remote SC226x EV1527"
Registering protocol [19] "TFA-Twin-Plus-30.3049 and Ea2 BL999"
Registering protocol [20] "Fine Offset Electronics WH1080/WH3080 Weather Station"
Registering protocol [21] "WT450"
Registering protocol [22] "LaCrosse WS-2310 Weather Station"
Registering protocol [23] "Esperanza EWS"
Registering protocol [24] "Generic temperature sensor 1"
Registering protocol [25] "WG-PB12V1"
Registering protocol [26] "HIDEKI TS04 Temperature, Humidity, Wind and Rain Sensor"
Registering protocol [27] "Watchman Sonic / Apollo Ultrasonic / Beckett Rocket oil tank monitor"
Registering protocol [28] "CurrentCost Current Sensor"
Registering protocol [29] "emonTx OpenEnergyMonitor"
Registering protocol [30] "HT680 Remote control"
Registering protocol [31] "S3318P Temperature & Humidity Sensor"
Registering protocol [32] "Akhan 100F14 remote keyless entry"
Registering protocol [33] "Quhwa"

pi@raspberrypi:~/rtl_433_rcswitch/build/src $ ./rtl_433
Registering protocol [1] "RC-Switch Remote Control"
Registering protocol [2] "Rubicson Temperature Sensor"
Registering protocol [3] "Prologue Temperature Sensor"
Registering protocol [4] "Waveman Switch Transmitter"
Registering protocol [5] "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol [6] "Acurite 609TXC Temperature and Humidity Sensor"
Registering protocol [7] "Oregon Scientific Weather Sensor"
Registering protocol [8] "Mebus 433"
Registering protocol [9] "KlikAanKlikUit Wireless Switch"
Registering protocol [10] "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol [11] "Fine Offset Electronics, WH2 Temperature/Humidity Sensor"
[... 54 lines cut ...]
Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
pulse_FSK_detect(): Maximum number of pulses reached!
2017-05-20 17:42:12 :   RC-Switch       :       05fdab80
2017-05-20 17:42:15 :   RC-Switch       :       05fdae80
2017-05-20 17:42:16 :   RC-Switch       :       05fdab80
2017-05-20 17:42:17 :   RC-Switch       :       05fdae80
2017-05-20 17:42:20 :   RC-Switch       :       1d706d80
2017-05-20 17:42:21 :   RC-Switch       :       05fdae80
2017-05-20 17:42:21 :   RC-Switch       :       05fdad80
2017-05-20 17:42:22 :   RC-Switch       :       05fdab80
2017-05-20 17:42:23 :   RC-Switch       :       05fdae80
2017-05-20 17:42:25 :   RC-Switch       :       1d706d80
2017-05-20 17:42:26 :   RC-Switch       :       05fdad80
2017-05-20 17:42:27 :   RC-Switch       :       05fdab80
!!

Super BIG Friends!!! Make general repository!!

zuckschwerdt, 2017-05-20:
Please tell me what `RC-Switch : 05fdae80`, `RC-Switch : 1d706d80, ... mean?
What button did you press? Is it button 1, 2, 3, 4, each with ON and OFF?
Also what is the brand and name, type of the RC-Switch?

directman66, 2017-05-20:
I have many different remotes for switches, I have pressed different buttons

directman66, 2017-05-20:
It seems that your application is taking raspi to reboot

zuckschwerdt, 2017-05-20:
Unlikely. All new code for the RC-Switch is this: https://github.com/zuckschwerdt/rtl_433/commit/f43f722b20f024c919025dda2500aec9ecd03de5

zuckschwerdt, 2017-05-20:
Most likely cause for a sudden reboot is undervoltage. Your power supply maybe too weak, the rtl receiver will use 0.4W in standby and 1.4W in operation. The Raspi will take ~4W on it's own when using all cores. Add a few devices and you will overload a common 10W USB charger.

directman66, 2017-05-20:
Our community actively uses your application, but there is not enough support for Russian-speaking users. I think very many users need this code. Our community is users of the smart house system majordomo. Ideally, from your application, get the plug-in for this system. I already plan to study how this can be done.

directman66, 2017-05-20:
http://majordomohome.com

directman66, 2017-05-20:
So please transfer the code to the main repository))

zuckschwerdt, 2017-05-20:
I would like to decode this protocol properly first. Please use every remote and press every button and then make a list:

- Remote name, Button name, output code

directman66, 2017-05-20:
Well, I'll try all my consoles and let you know

directman66, 2017-05-20:
Also on this protocol are many pir sensors of security systems

zuckschwerdt, 2017-05-20:
If you get strange output thats not from your own devices: try to guess what it could be. It will help to figure out the general protocol.

Also not to give the wrong impression, I'm not "the rtl_433 guy", that would be Ben. I'm just another interested user of rtl_433 ;)

directman66, 2017-05-21:
[image]
[image]

directman66, 2017-05-21:
[image]
[image]

directman66, 2017-05-21:
[image]
[image]

directman66, 2017-05-21:
[image]
[image]

no decode

directman66, 2017-05-21:
old mail aplication 
[image]
[image]

directman66, 2017-05-21:
in new aplication code fffffff80 for all button

zuckschwerdt, 2017-05-21:
Impressive collection! A real smart home. If the code show FFFFFFFF then the timing must be wrong. All bits are seen as `1`. Can you record another sample?
If you use `rtl_433 -a -t` you will get one file for each button press. That will be best to sort through.

directman66, 2017-05-21:
[image]
[image]
[image]

button 1,2,3

directman66, 2017-05-21:
button 1
[image]
[image]
[image]

directman66, 2017-05-21:
BUTTON 2
[image]
[image]

zuckschwerdt, 2017-05-21:
Ok, nice. So we got BB37 as sync word or maybe ID and the last nibble encodes the button. Also the trailing bit doesn't seem needed.
Does your remote have a House-Code or Channel settings? Like this: https://en.wikipedia.org/wiki/DIP_switch
Maybe take a picture of the remote's backside.

The code is updated. Load & compile it again.

directman66, 2017-05-21:
dip switch nothing

directman66, 2017-05-21:
[image]

nothing decode

zuckschwerdt, 2017-05-21:
Send me the `gfile0xx.data`  so I can have a look.

directman66, 2017-05-21:
https://yadi.sk/mail/?hash=q8WGrpYwiQ7PR1Ox%2B5keA6gdFvQt8QnNPg3rkm%2F6f6o%3D

directman66, 2017-05-21:
sory, cu8 only

zuckschwerdt, 2017-05-21:
I already have that cu8 file. I made the decoder using that file :) I need the gfile*.data you made with `rtl_433 -a -t`. It is in the pictures above as `*** Saving signal to file gfil052.data`. Look in den `build` or `build/src` directory.

directman66, 2017-05-21:
Good friend, I can only in a few days

zuckschwerdt, 2017-05-21:
Don't worry. Whenever you are ready, I'll help you :)

merbanan, 2018-12-03:
No activity for a while. Closing issue.

closed 2018-12-03 by merbanan

kueblc, 2019-02-09:
Hey @zuckschwerdt, just curious if you still have the feat-rcswitch branch somewhere, or if this was merged. I too am trying to integrate 433mhz wall switches with my home automation setup.

So far I've only been able to capture a burst of three packets for each activation but not decode the contents. I tried my hand at manually decoding the capture with GQRX and theorize it's 64-65 bits (or maybe half that if the bits are encoded by pairs) FSK encoded, but I'm fairly new to this field so I could be completely wrong. I also found that the resolution is really being pushed to the limit when using GQRX so it may be that my SDR isn't going to work for this, but coming across this issue I have renewed hope.

Thanks for any insight you might have.

zuckschwerdt, 2019-02-09:
Much happened since then. Read https://github.com/merbanan/rtl_433/wiki/Adding-a-new-remote-device to get a brief guide how to figure signals out.
Switches, doorbells, PIR motion, etc… are now supported with a rather simple config syntax. See e.g. https://github.com/merbanan/rtl_433/blob/master/conf/silverline_doorbell.conf

kueblc, 2019-02-09:
Hi @zuckschwerdt, thanks for the quick feedback, I really appreciate it!

I have tried following those guides before but as I don't have a background in RF, and the guide doesn't really delve into "how" to differentiate different RF signal types, I'm not sure if I'm doing something wrong or if my hardware is just plain incompatible.

The main problem I seem to be having is that while I can detect 3 bursts of activity, the data within those bursts isn't discernible. Thus I can create a fairly "dumb" detection system that knows that a button is pushed but not "which one".

I had already tried GQRX and Audacity, though it isn't clear from the guide what I need to be looking for in a FSK style transmission. I tried PulseView for the first time today and the UI is much more obvious, but nothing shows on the FSK line and the ASK line only shows a single block for each packet rather than individual bits. The FM line seems to be the most promising, as it looks a little more like organized data than the others.

I'll attach the capture here on the off chance you or someone more experienced could take a look. Once I get a better grasp of what I'm doing I'd love to contribute back to the project, improving documentation and/or code. Thanks again for your time.

[g001_433.92M_1024k.zip](https://github.com/merbanan/rtl_433/files/2848229/g001_433.92M_1024k.zip)

kueblc, 2019-02-09:
And here is what I found via PulseView

[image]

merbanan, 2019-02-10:
This is a fsk signal. Unfortunately the demodulator does not output any bits from the signal. It might be because the signal is to strong. There are signs in the I view of clipping.

kueblc, 2019-02-10:
Ah I hadn't even considered it might be clipping. On earlier attempts I had suspected it was too weak and moved the equipment as close as possible for testing. Is there hope yet that I might extract data from this via rtl_433 if the gain or physical distance is adjusted?

merbanan, 2019-02-10:
Sure, might take some developer effort but we usually get there.

kueblc, 2019-02-10:
I'm thrilled to hear it! Thanks for the very prompt response. Anything I should be doing to help?

> Unfortunately the demodulator does not output any bits from the signal

To clarify, is the demodulator in rtl_433 capable of handling this style of transmission? Or are you talking about PulseView?

merbanan, 2019-02-10:
If you look at the bottom FM view you see the output of the rtl_433 FM demodulator. What fails is the bit recovery. That might work better with a non-clipped signal. If not one need to tune parts of the bit recovery code for proper operation.

kueblc, 2019-02-10:
Understood. Thank you very much for the insight, I really appreciate it! I'll do some more digging on this later today and see if I can get a cleaner capture.

kueblc, 2019-02-10:
Seem to be getting somewhere! I can see stuff happening in the FSK row in PulseView.

[image]

I'm not really sure where to go from here, please forgive my inexperience with RF. I'm reading the guide on the wiki linked above, it says to measure short and long pulses but I'm seeing all sorts of pulse lengths.