merbanan/rtl_433 #549: For some reason, the value of the ID field changes. I can not find the reason.

directman66, 2017-05-22:
For some reason, the value of the ID field changes over time. I thought that for one sensor there should be a constant some field, nevertheless I got about 100 names of sensors based on this field. Why does the field go changing with time? Thank you.

{"time" : "2017-05-22 10:13:01", "model" : "Nexus Temperature/Humidity", "id" : 194, "battery" : "LOW", "channel" : 1, "temperature_C" : 6.500, "humidity" : 103}

{"time" : "2017-05-22 13:14:44", "model" : "Nexus Temperature/Humidity", "id" : 195, "battery" : "LOW", "channel" : 1, "temperature_C" : 6.700, "humidity" : 103}

directman66, 2017-05-22:
[image]

So many objects from id

rct, 2017-05-22:
That decoding looks wrong, unless your humidity is very high and your temperature is very cold. 

Do you own "Nexus" sensors?    Or is this a false positive that this decoder is running for some other signal?

I'm getting false positive decodes for LaCrosse TX141.  I don't own any of those sensors.  I don't believe I'm receiving a nearby sensor from someone else because the temperatures are implausible.

If you don't have Nexus sensors, you can enable only the decoders you actually need.

zuckschwerdt, 2017-05-22:
The original request was vor "Auriol-v2", false positive seems likely. But supporting the new protocol would require a number of readings paired with manual measured data (best case: taken directly from the display unit). Also we would need a clean gfile to maybe spot timing differences.

zuckschwerdt, 2017-05-23:
The signal seems almsot identical. This needs to be figured out looking at the payload.
Original Nexus:
```
Total count:  444,  width: 229171		(916.7 ms)
Pulse width distribution:
 [ 0] count:  443,  width:   124 [121;139]	( 496 us)
 [ 1] count:    1,  width:    34 [34;34]	( 136 us)
Gap width distribution:
 [ 0] count:  240,  width:   486 [482;598]	(1944 us)
 [ 1] count:  191,  width:   241 [237;247]	( 964 us)
 [ 2] count:   11,  width:   974 [971;977]	(3896 us)
 [ 3] count:    1,  width:   358 [358;358]	(1432 us)
Pulse period distribution:
 [ 0] count:  240,  width:   610 [606;720]	(2440 us)
 [ 1] count:  191,  width:   366 [362;372]	(1464 us)
 [ 2] count:   11,  width:  1099 [1098;1100]	(4396 us)
 [ 3] count:    1,  width:   481 [481;481]	(1924 us)
```
Auriol-v2:
```
Total count:  444,  width: 220124		(880.5 ms)
Pulse width distribution:
 [ 0] count:    1,  width:    40 [40;40]	( 160 us)
 [ 1] count:  443,  width:   127 [125;132]	( 508 us)
Gap width distribution:
 [ 0] count:   12,  width:  1009 [1007;1011]	(4036 us)
 [ 1] count:  180,  width:   498 [494;501]	(1992 us)
 [ 2] count:  251,  width:   246 [242;249]	( 984 us)
Pulse period distribution:
 [ 0] count:   12,  width:  1129 [1050;1137]	(4516 us)
 [ 1] count:  180,  width:   625 [624;628]	(2500 us)
 [ 2] count:  251,  width:   373 [372;375]	(1492 us)
```

merbanan, 2018-12-03:
Sample signals are missing, closing issue. The 36 bit OOK based protocols all collide one in a while. Might be possible to keep them a part by looking at amount of repeats etc.

closed 2018-12-03 by merbanan