merbanan/rtl_433 #556: efergy_e2_classic.c: undefined reference to `pow'

killerbees19, 2017-05-29:
Can't compile at Debian Jessie:

```
libtool: link: gcc -std=gnu99 -g -O2 -I/usr/include/ -I/usr/include/libusb-1.0 -Wall -Wextra -Wno-unused -Wsign-compare -fPIC -fvisibility=hidden -g -O2 -I/usr/include/ -I/usr/include/libusb-1.0 -Wall -Wextra -Wno-unused -Wsign-compare -o rtl_433 baseband.o bitbuffer.o data.o pulse_demod.o pulse_detect.o rtl_433.o util.o devices/acurite.o devices/alecto.o devices/ambient_weather.o devices/blyss.o devices/brennenstuhl_rcs_2044.o devices/calibeur.o devices/cardin.o devices/current_cost.o devices/chuango.o devices/danfoss.o devices/dsc.o devices/ec3k.o devices/efergy_e2_classic.o devices/efergy_optical.o devices/elro_db286a.o devices/elv.o devices/emontx.o devices/esperanza_ews.o devices/fineoffset.o devices/fineoffset_wh1080.o devices/generic_remote.o devices/generic_temperature_sensor.o devices/gt_wt_02.o devices/hideki.o devices/hondaremote.o devices/ht680.o devices/inovalley-kw9015b.o devices/intertechno.o devices/kedsum.o devices/lacrosse.o devices/lacrosse_TX141TH_Bv2.o devices/lacrossews.o devices/lacrosse_tx35.o devices/lightwave_rf.o devices/mebus.o devices/newkaku.o devices/nexus.o devices/oil_watchman.o devices/oregon_scientific.o devices/oregon_scientific_v1.o devices/prologue.o devices/quhwa.o devices/rubicson.o devices/silvercrest.o devices/springfield.o devices/steffen.o devices/tfa_twin_plus_30.3049.o devices/tfa_pool_thermometer.o devices/valeo.o devices/waveman.o devices/wt450.o devices/x10_rf.o devices/s3318p.o devices/akhan_100F14.o devices/proove.o devices/bresser_3ch.o devices/oregon_scientific_sl109h.o devices/steelmate.o devices/schraeder.o devices/new_template.o devices/radiohead_ask.o devices/kerui.o devices/fineoffset_wh1050.o devices/honeywell.o devices/maverick_et73x.o devices/rftech.o devices/vaillant_vrt340f.o devices/wg_pb12v1.o devices/ibis_beacon.o devices/oil_standard.o devices/tpms_citroen.o  -lrtlsdr -lusb-1.0
devices/efergy_e2_classic.o: In function `efergy_e2_classic_callback':
/foo/bar/rtl_433/src/devices/efergy_e2_classic.c:75: undefined reference to `pow'
```

Simple workaround at the moment:

```sh
cd src && gcc -std=gnu99 -g -O2 -I/usr/include/ -I/usr/include/libusb-1.0 -Wall -Wextra -Wno-unused -Wsign-compare -fPIC -fvisibility=hidden -g -O2 -I/usr/include/ -I/usr/include/libusb-1.0 -Wall -Wextra -Wno-unused -Wsign-compare -o rtl_433 baseband.o bitbuffer.o data.o pulse_demod.o pulse_detect.o rtl_433.o util.o devices/acurite.o devices/alecto.o devices/ambient_weather.o devices/blyss.o devices/brennenstuhl_rcs_2044.o devices/calibeur.o devices/cardin.o devices/current_cost.o devices/chuango.o devices/danfoss.o devices/dsc.o devices/ec3k.o devices/efergy_e2_classic.o devices/efergy_optical.o devices/elro_db286a.o devices/elv.o devices/emontx.o devices/esperanza_ews.o devices/fineoffset.o devices/fineoffset_wh1080.o devices/generic_remote.o devices/generic_temperature_sensor.o devices/gt_wt_02.o devices/hideki.o devices/hondaremote.o devices/ht680.o devices/inovalley-kw9015b.o devices/intertechno.o devices/kedsum.o devices/lacrosse.o devices/lacrosse_TX141TH_Bv2.o devices/lacrossews.o devices/lacrosse_tx35.o devices/lightwave_rf.o devices/mebus.o devices/newkaku.o devices/nexus.o devices/oil_watchman.o devices/oregon_scientific.o devices/oregon_scientific_v1.o devices/prologue.o devices/quhwa.o devices/rubicson.o devices/silvercrest.o devices/springfield.o devices/steffen.o devices/tfa_twin_plus_30.3049.o devices/tfa_pool_thermometer.o devices/valeo.o devices/waveman.o devices/wt450.o devices/x10_rf.o devices/s3318p.o devices/akhan_100F14.o devices/proove.o devices/bresser_3ch.o devices/oregon_scientific_sl109h.o devices/steelmate.o devices/schraeder.o devices/new_template.o devices/radiohead_ask.o devices/kerui.o devices/fineoffset_wh1050.o devices/honeywell.o devices/maverick_et73x.o devices/rftech.o devices/vaillant_vrt340f.o devices/wg_pb12v1.o devices/ibis_beacon.o devices/oil_standard.o devices/tpms_citroen.o  -lrtlsdr -lusb-1.0 -lm && cd .. && V=99 make
```

`-lm` fixed it for me.

zuckschwerdt, 2017-05-29:
This due to https://github.com/merbanan/rtl_433/pull/540/files#diff-dd2a3b66425f9817f191789cc9de632fR75
Perhaps pre-scaling and some integer bit-shifting could replace that.

DavidBel86, 2017-05-29:
Hi I adapted the code, I removed the pow and replaced by a division and bit shifting on my repo.
I'll do a pull request ;)

DavidBel86, 2017-05-30:
PR done, should be ok, could you test?

killerbees19, 2017-05-30:
Confirmed, works! :-)

closed 2017-05-30 by killerbees19