merbanan/rtl_433 #558: Efergy creates invalid JSON

azrdev, 2017-06-02:
I've got the following output from `rtl_433 -F json`:
~~~
{"time" : "2017-06-02 12:09:05", "model" : "Efergy Optical", "pulses" : 3200, "energy" : -nan}
{"time" : "2017-06-02 12:09:05", "model" : "Efergy Optical", "pulses" : 2000, "energy" : -nan}
{"time" : "2017-06-02 12:09:05", "model" : "Efergy Optical", "pulses" : 1000, "energy" : -nan}
{"time" : "2017-06-02 12:09:05", "model" : "Efergy Optical", "pulses" : 500, "energy" : -nan}
~~~
Which is not valid JSON (see e.g. <https://stackoverflow.com/questions/6601812/sending-nan-in-json>)

merbanan, 2017-06-02:
Supply a sample signal that trigger this decode.

rct, 2017-06-04:
Seems line there are two issues here:

1. Why is the efergy optical decoder outputting NaNs?   (It doesn't look like there is any valid data in that message, so is that a false positive or other error condition that should be caught and ignored earlier?)
2. If NaN isn't valid in JSON, data_make() should catch that and output Null or whatever is technically correct for JSON, or alternatively omit the field.  There maybe subtleties of working implementation/stand practices here, though a quick search yields [ECMA-404](https://www.ecma-international.org/publications/standards/Ecma-404.htm) as a standard, which says a number can't have non-numeric values like NaN or Infinity.  Null appears to be acceptable (though it's not a number, it's a type.)  Most likely a separate (github) issue should be opened for this.

zuckschwerdt, 2017-07-07:
The problem seems to be with the assumption that a CRC will guarantee a valid packet. A packet of all zeros needs to be excluded since the CRC will return the `init` which is zero here and thus compares to valid. In turn this gives a `seconds` value of zero which is used as divsor, hence NaN.
I'll supply a PR shortly.

zuckschwerdt, 2017-07-07:
Note that `seconds` is set to the wrong byte (the crc) and that way the bug is triggerd for ~1 in 256 messages. Also the readings with more than 3 pulses per 15 seconds are garbage that way.

merbanan, 2018-12-03:
Is this issue still valid ?

zuckschwerdt, 2018-12-04:
Already fixed.

closed 2018-12-04 by zuckschwerdt