merbanan/rtl_433 #559: Oregon Scientific THGN132N

killerbees19, 2017-06-04:
Is that sensor already compatible? It's v2.1 and should be nearly identical to THGR122NX. Not sure if it's the same as THGR122N, without trailing X in the name, that I've found in the source code.

killerbees19, 2017-06-05:
Maybe I'll buy a set of two `THGN132N` or `THGR122NX`. I'll think about it.

If they are not working OOTB, and there is no simple solution, could somebody help me to implement the required changes? I'm not an expert in C programming and I think the decoding stuff is exceeding my skills…

zuckschwerdt, 2017-06-06:
If you record a few clean signals I guess we could come up with something ;)

killerbees19, 2017-06-06:
I've ordered only one THGR122NX for the moment. I'll arrive between 9th and 12th. Can't wait to see the raw data... 😁

killerbees19, 2017-06-10:
According to [OregonScientific-RF-Protocols.pdf](http://wmrx00.sourceforge.net/Arduino/OregonScientific-RF-Protocols.pdf) (and the code in oregon_scientific.c), `0x1d20` should be the code for `THGR122N` and `THGR122NX`. So far, no luck with device "12"…

**THGR122NX**: 25° / 47% (channel 3)
`~/bin/rtl_433 -l 1 -g 50 -R 12 -D`

```
Badly formatted OS v2.1 message encountered.
3d ff ff ff fb df de fd 5b be f7 ff fb bb 7f a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Unrecognized Msg in v3: 3d ff ff ff fb df de fd 5b be f7 ff fb bb 7f a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Badly formatted OS v2.1 message encountered.
5d b5 ff ff ed ff fd df ff ef ef fe b6 b7 ff fd c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Unrecognized Msg in v3: 5d b5 ff ff ed ff fd df ff ef ef fe b6 b7 ff fd c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

------------------------------------------------------------------------

Badly formatted OS v2.1 message encountered.
7e df be ff ff ff ff fe ff ff ff ff ff ff e0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Unrecognized Msg in v3: 7e df be ff ff ff ff fe ff ff ff ff ff ff e0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

------------------------------------------------------------------------

Possible Oregon Scientific v2.1 message, but sync nibble wasn't found
Raw Data: 55 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Badly formatted OS v2.1 message encountered.
7f fd ff df fd ff ff 57 ff fb bd fa aa 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Unrecognized Msg in v3: 7f fd ff df fd ff ff 57 ff fb bd fa aa 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Badly formatted OS v2.1 message encountered.
7f 7f ff ff bf ff fe ff 5f ff eb ff c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Unrecognized Msg in v3: 7f 7f ff ff bf ff fe ff 5f ff eb ff c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

I think the output of `-a/-A` or a `gqrx` record isn't required now?

zuckschwerdt, 2017-06-10:
If you could could capture a few samples that would be great. Perhaps have a look with https://github.com/zuckschwerdt/sdr-spectrogram if the capture data has a clean spectrogram.

killerbees19, 2017-06-10:
> If you could could capture a few samples that would be great.

You mean recording with `gqrx` into a `*.raw` file?

killerbees19, 2017-06-10:
5 recordings, the newest two files with other antenna: https://happytec.at/r/Qfaab

zuckschwerdt, 2017-06-10:
gqrx records as complex float, to convert to rtl_433 compatible file use:
```
sox -t raw -b 32 -c 2 -e float -r 250000 file.cf32 -t raw -b 8 -c 2 -e unsigned-integer file.cu8
```
(gqrx `.raw` is `.cf32` and rtl_433 `.data` is `.cu8`). I'll have a peek into the samples shortly.

zuckschwerdt, 2017-06-10:
What's the samplerate here? It's PCM with manchester encoding but 890 samples per pulse (half-bit rate for manchester encoding) looks more like a 1024k samplerate than the usual 250k.

zuckschwerdt, 2017-06-10:
Somewhat funny signal. Transcribing the pulses (first capture, first packet) we get
```
11001100110011001100110011001100110011001100110011001100110011010010110100101100110100110011001011010010110011010010110100110011001100110011001100110010110100110010110011010010110011001100110100110011001100110011001100101100110100101101001100101101001100110011001100110010110011001101001100110010110100110011001100110010110011010011001100110010110100101101001011010011001100110011001
```

Manchester decoded that's
```
01010101010101010101010101010100110011010010101100110100110010101010101010110010110100110101010010101010101011010011001011001010101010110101001010110010101010110100101010110011001100101010101
= 0x2aaaaaaa66959a65555969aa5556996555a95955a5599955
```

Looks suspicious? Yeah, it's again Manchester encoded. Decoding now yields
```
0x7fff 0xa74bfd90fe5bf1df3d5f
```
A clean preamble of 1-bits and a nice 10 byte payload.

zuckschwerdt, 2017-06-10:
If I assume a samplerate of 2048k it works with the current THGR122N code.
```
rtl_433 -s 2048000 -q -F json -r gqrx_20170610_132925_433920000_1800000_fc.cu8
Test mode active. Reading samples from file: gqrx_20170610_132925_433920000_1800000_fc.cu8
{"time" : "@1.376000s", "brand" : "OS", "model" : "THGR122N", "id" : 246, "channel" : 3, "battery" : "OK", "temperature_C" : 25.800, "humidity" : 47}
{"time" : "@1.568000s", "brand" : "OS", "model" : "THGR122N", "id" : 246, "channel" : 3, "battery" : "OK", "temperature_C" : 25.800, "humidity" : 47}
```

zuckschwerdt, 2017-06-10:
Well, if I'd only bothered to read the OS protocol notes. The second layer of encoding is expected (the notes talk about interleaving inverted bits, I'd call it Manchester coding…). So, no mystery here.

killerbees19, 2017-06-10:
Indeed, it works with 2048k sample rate:

```
Found Fitipower FC0013 tuner
Sample rate set to 2048000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Tuned to 433920000 Hz.
{"time" : "2017-06-10 18:30:45", "brand" : "OS", "model" : "THGR122N", "id" : 246, "channel" : 3, "battery" : "OK", "temperature_C" : 26.900, "humidity" : 40}
{"time" : "2017-06-10 18:30:45", "brand" : "OS", "model" : "THGR122N", "id" : 246, "channel" : 3, "battery" : "OK", "temperature_C" : 26.900, "humidity" : 40}
{"time" : "2017-06-10 18:31:28", "brand" : "OS", "model" : "THGR122N", "id" : 246, "channel" : 3, "battery" : "OK", "temperature_C" : 26.900, "humidity" : 40}
{"time" : "2017-06-10 18:31:28", "brand" : "OS", "model" : "THGR122N", "id" : 246, "channel" : 3, "battery" : "OK", "temperature_C" : 26.900, "humidity" : 40}
```

But now I'm unable to receive any other sensor data… :-)

killerbees19, 2017-06-10:
So this means it's impossible to fetch the THGR122NX and any other sensor at the same time with only one DVB-T stick? This one needs 2048k sample rate but encoding of other devices fails at everything higher than 250k. Hmmm, any ideas?

zuckschwerdt, 2017-06-10:
I wouldn't say that it needs 2048k, you just happend to record from gqrx with 2048k. The signal isn't very fast 250k should be fine. The frequency in the samples seems close to 433.92 -- a higher samplerate is only needed if the frequency is off to much (bandwidth = samplerate with I/Q data, i.e. no aliasing). The captured filenames `gqrx_20170610_132925_433920000_1800000_fc.raw` seem to suggest a 1800k samplerate? Perhaps the signal timing is different? Maybe record from rtl_433 with `-a -t` switches.

killerbees19, 2017-06-10:
Ok, new try: https://happytec.at/r/Rfaab

But there are many sensors in the air, not only mine: OSv1, Acurite, Nexus, raingauges, …

killerbees19, 2017-06-10:
And yes, you're right. gqrx recorded with 1800k...

I forgot the default values at the initial device configuration dialog at first startup. Stupid me! :-(

killerbees19, 2017-06-10:
btw: [Here](https://happytec.at/r/Sfaab) is another sample from gqrx, now with 250k and already converted with sox… :-)

zuckschwerdt, 2017-06-11:
Measuring the signal shows the same exact pulse width used in other OS devices (488µs). You should be good just running current rtl_433. The capture shows (`rtl_433 -f 433920000 -s 250000 -q -F json -r $file`):
```
gfile002.data
{"time" : "@0.000000s", "brand" : "OS", "model" : "THGR122N", "id" : 246, "channel" : 3, "battery" : "OK", "temperature_C" : 26.100, "humidity" : 42}
{"time" : "@0.262144s", "brand" : "OS", "model" : "THGR122N", "id" : 246, "channel" : 3, "battery" : "OK", "temperature_C" : 26.100, "humidity" : 42}

gfile016.data
{"time" : "@0.262144s", "model" : "OSv1 Temperature Sensor", "sid" : 0, "channel" : 1, "battery" : "OK", "temperature_C" : 25.300}
{"time" : "@0.524288s", "model" : "OSv1 Temperature Sensor", "sid" : 0, "channel" : 1, "battery" : "OK", "temperature_C" : 25.300}

gfile021.data
{"time" : "@1.310720s", "brand" : "OS", "model" : "THGR122N", "id" : 246, "channel" : 3, "battery" : "OK", "temperature_C" : 26.100, "humidity" : 42}
{"time" : "@1.572864s", "brand" : "OS", "model" : "THGR122N", "id" : 246, "channel" : 3, "battery" : "OK", "temperature_C" : 26.100, "humidity" : 42}

gfile029.data
{"time" : "@1.572864s", "model" : "OSv1 Temperature Sensor", "sid" : 0, "channel" : 1, "battery" : "OK", "temperature_C" : 25.300}
{"time" : "@1.835008s", "model" : "OSv1 Temperature Sensor", "sid" : 0, "channel" : 1, "battery" : "OK", "temperature_C" : 25.300}
```

killerbees19, 2017-06-11:
Interesting, because I had no luck to decode it in realtime without analyse mode. I can see all other sensors, but not the THGR122NX.

I'll test it again later... :-/

zuckschwerdt, 2017-06-11:
Can you confirm the offline decoding of gfile002.data and gfile021.data?

killerbees19, 2017-06-11:
Update: I'm playing around with the bit detection level (again). Currently 100% success in realtime for THGR122N, 2 × THN128, THR128, Acurite and Nexus devices.

We'll see if it's stable now: `-g 50 -l 10`

killerbees19, 2017-06-11:
Update: Works great in the next room, but still no success in the same room. The THGR122NX signal is "too strong". A THN128 in the same room works without any problems...

Furthermore I need a gain of >=25 to receive my OSv1 data...

Current args: `-g 25 -l 10`

killerbees19, 2017-06-14:
Looks like it's pure finetuning of -g/-l. It's still not perfect, but I'm closing this issue now...

[image]
[image]

Thanks for your help! :-)

closed 2017-06-14 by killerbees19

killerbees19, 2022-01-13:
For the record: `-Y autolevel` fixed all issues.

Thanks! Nice feature 😍

zuckschwerdt, 2022-01-13:
Thanks for the feedback, we'll likely make that a default soon.