merbanan/rtl_433 #563: Solight TE44

oujesky, 2017-06-11:
I have mostly decoded my home wireless thermometer, it is some kind of generic simple chinese transmitter used in different kits, so far I have found two (Solight TE44 and EMOS E0107T) sold mostly in central europe.

---

I am able to decode the data discovered by `rtl_433 -a -t`, the format is following:

12 repetitions of the same 36 bit payload, 1 zero between each repetition.

36 bit payload format: `xxxxxxxx 10ccmmmm tttttttt 1111hhhh hhhh`

`x` - random key - changes after device reset - 8 bits
`c` - channel (0-2) - 2 bits
`m` - multiplier - signed integer, two's complement - 4 bits
`t` - temperature in celsius - unsigned integer - 8 bits
`h` - checksum - 8 bits

temperature = `((256 * m) + t) / 10`

(not sure how the checksum is calculated yet but it seems it takes at least the first three bytes into account)

---

Sample output from `rtl_433 -r gfile022.data -a` (attached [gfile022.zip](https://github.com/merbanan/rtl_433/files/1066479/gfile022.zip)):

```
Test mode active. Reading samples from file: gfile022.data
Input format: uint8
*** signal_start = 16227, signal_end = 262145
signal_len = 245918,  pulses = 455
Iteration 1. t: 126    min: 125 (94)    max: 127 (361)    delta 4
Iteration 2. t: 126    min: 125 (1)    max: 127 (454)    delta 0
Distance coding: Pulse length 126

Short distance: 233, long distance: 483, packet distance: 971

p_limit: 126
bitbuffer:: Number of rows: 12 
[00] {37} 94 91 15 fe 90 : 10010100 10010001 00010101 11111110 10010
[01] {37} 94 91 15 fe 90 : 10010100 10010001 00010101 11111110 10010
[02] {37} 94 91 15 fe 90 : 10010100 10010001 00010101 11111110 10010
[03] {37} 94 91 15 fe 90 : 10010100 10010001 00010101 11111110 10010
[04] {37} 94 91 15 fe 90 : 10010100 10010001 00010101 11111110 10010
[05] {37} 94 91 15 fe 90 : 10010100 10010001 00010101 11111110 10010
[06] {37} 94 91 15 fe 90 : 10010100 10010001 00010101 11111110 10010
[07] {37} 94 91 15 fe 90 : 10010100 10010001 00010101 11111110 10010
[08] {37} 94 91 15 fe 90 : 10010100 10010001 00010101 11111110 10010
[09] {37} 94 91 15 fe 90 : 10010100 10010001 00010101 11111110 10010
[10] {37} 94 91 15 fe 90 : 10010100 10010001 00010101 11111110 10010
[11] {36} 94 91 15 fe 90 : 10010100 10010001 00010101 11111110 1001
Test mode file issued 4 packets
```
decoded as:

channel 2 (`c` = 1, indexed from 0)
`m` = 1 
`t` = 21

temperature 27,7 C - ((256 * 1) + 21) / 10

Output from `rtl_433 -r gfile022.data -A` does not make much sense.

---

My question is, what settings should I specify in `r_device` struct to get any sensible data to my callback? So far I was not able to figure anything out, even though I experimented quite a lot with those settings.

Could anybody help me with finding the optimal settings for `modulation`, `short_limit`, `long_limit`, `reset_limit`? Thanks

zuckschwerdt, 2017-06-11:
Looks like either Return-to-Zero (RZ) encoding or Pulse Position Modulation. Your decoding above assumes the later.
Short gap is 972 us, long gap is 1932 us, packet gap is 3880 us
Try `OOK_PULSE_PPM_RAW` with
```
short_limit:  972*1.5 // Threshold between short and long gap [us]
long_limit:  1932*1.5 // Maximum gap size before new row of bits [us]
reset_limit: 3880*1.5 // Maximum gap size before End Of Message [us]
```

oujesky, 2017-06-11:
Thank you, these settings works like charm.

PR #564 (2017-06-15): Solight TE44 wireless thermometer
Solight TE44 

Closes #563. I will add testing data to https://github.com/oujesky/rtl_433_tests too.

[image]
closed 2017-06-15 by merbanan