merbanan/rtl_433 #567: LaCrosse TX Sensor duplicate demod/output

pclov3r, 2017-06-17:
It appears that during the demodulation for LaCrosse TX Sensors it duplicates the entry.

Example output:

2017-06-17 00:32:58 :	LaCrosse TX Sensor 	:	50
	Temperature:	 18.9 C
2017-06-17 00:32:58 :	LaCrosse TX Sensor 	:	50
	Temperature:	 18.9 C
2017-06-17 00:32:58 :	LaCrosse TX Sensor 	:	50
	Humidity:	 68.0 %
2017-06-17 00:32:58 :	LaCrosse TX Sensor 	:	50
	Humidity:	 68.0 %

zuckschwerdt, 2017-06-17:
Is this on a single transmitted packet? Do you have a capture (`-a -t` flags) or can you aquire a capture?
Maybe it's just a repeated transmission, then you have to filter this at a later stage (mostly with custom logging).

pclov3r, 2017-06-17:
I am using a TX7U sensor. 

The sensor will send temperature and then immediately followed by humidity.

Before this was a non issue and it worked correctly. Rebuilt rtl_433 and this behavior is now occurring so some commit i figure has caused by but unsure which.

pclov3r, 2017-06-17:
I did a capture and see this

p_limit: 225
bitbuffer:: Number of rows: 2
[00] {44} f5 f9 a9 76 97 50 : 11110101 11111001 10101001 01110110 10010111 0101
[01] {44} f5 f9 a9 76 97 50 : 11110101 11111001 10101001 01110110 10010111 0101
*** signal_start = 4367534, signal_end = 4438760
signal_len = 71226,  pulses = 88
Iteration 1. t: 225    min: 132 (32)    max: 318 (56)    delta 1
Iteration 2. t: 225    min: 132 (32)    max: 318 (56)    delta 0
Pulse coding: Short pulse length 132 - Long pulse length 318

Short distance: 239, long distance: 254, packet distance: 7544

p_limit: 225
bitbuffer:: Number of rows: 2
[00] {44} f5 19 a8 ff 8f e0 : 11110101 00011001 10101000 11111111 10001111 1110
[01] {44} f5 19 a8 ff 8f e0 : 11110101 00011001 10101000 11111111 10001111 1110

So it appears for some reason it's being duplicated when clearly there is only one packet each for temp and humidity.

zuckschwerdt, 2017-06-17:
The samples `rtl_433_tests/tests/lacrosse/01/gfile001.data` and `rtl_433_tests/tests/lacrosse/01/gfile002.data` show the same behaviour. I'll see if I can fix this.

rct, 2017-06-17:
The message is actually repeated and what is being output is correct.  Please don't "fix" it.  Duplicate suppression, including keeping state of what has been heard should be the job of the downstream client.

rct, 2017-06-17:
Note the behavior did change a while back , the code that kept track of the what was last heard was removed.   It is now consistent with other sensors.

zuckschwerdt, 2017-06-17:
I agree, but this is in a single packet, not two separate transmissions. Multiple rows in the same transmission should not call data_make multiple times but rather confirm a clean reception. That's how most devices handle this.

rct, 2017-06-17:
The assumption that a "packet" (one bitbuffer) represents a single transmission from a single device works well in testing a single device but winds up being naive in practice based on my experience.  I have about a half dozen of these devices, and a whole bunch of others, overlaps and collisions occur. Also, the notion of a single transmission can change depending on the limits (and the demodulator)  so assumptions about getting all the repeats in a single packet fall down.

Some decoders rely on these assumptions which make them less robust than they need to be.

If the device implements robust enough error checking in the protocol, there is no need to apply assumptions for attempting to determine correctness.  (On the other hand If you don't have any error checking, sure, doing it statistically is all you can do, but I'm still of the opinion that's something for a consumer to figure out what its criteria is.)

For temperature and humidity sensors it might be OK to just wait for the next cycle and hope for the best.  For other devices, like security contacts, it's worth taking steps to try to capture 

Even a fairly trivial consumer, can do a simple string compare to see if the current line is a duplicate of the previous line.

In the meantime, letting my consumer(s) see and count the repeats let's me keep stats on how well I'm hearing each device.   I can't do that if messages are being suppressed. 

I maintain that keeping state, determine what is and what isn't a duplicate, applying additional checks to improve data quality should all be the domain of the consumer, not the radio layer.

zuckschwerdt, 2017-06-17:
I see your point. Let's break it down. Some devices will transmit a burst of identical messages to increase robustness. There is a known fixed (short) distance between repeats and the `reset_limit` is used to decide if that should end up as rows in a single bitbuffer.

E.g. the Bresser sensor will send 15 identical messages, the Nexus 12. I'd say it's too verbose to output all that. Requiring a number of identical rows is a good way for the decoder to confirm a valid reception. I was thinking about those kind of transmission.

In case of the LaCrosse with just 2 repeats a different approach is better. As you point out requiring two identical messages would block out every slightly disturbed transmission. Which is rather bad, especially with critical messages from security devices.

I would not consider it "keeping state" when looking at the rows of a single bitbuffer, though. Case in point: the GS558 smoke detector will transmit the exact same alarm code in learn mode, just with every second row being a "learn" pattern.
Also when the protocol defines a message transmission as a fixed number of repeats (as in the examples above) it is not intended as duplicates but a single reading. Otherwise I completly agree about never keeping state! I guess we just might be talking about different things here :)

On to improve things: I'd really like to catch weak signals too while also getting an indication about validity. How about a "confidence" field to pass "reception strength" on to the consumer? In absence of a CRC or checksum that would be the only way to decide on a reading. Normalized to e.g. percent preferably.

(I'm also working on an optional RSSI and SNR indicator for debugging and setting up gain/level which might be helpful too.)

TL;DR thanks for taking the time to explain, LaCrosse is good as-is. Might be worthwhile to explore validity/confidence stuff someday.

rct, 2017-06-17:
I'll come back to the discussion tomorrow, but I wanted to say, I'm very interested in getting some signal information (RSSI/SNR/noise floor, etc.) optionally output as part of the data stream.  I've struggled with determine what occasionally changes in my RF environment that impedes reception.  I'm in an inherently noisy RF area (New York City).   I've got saved data structured data going back to 2015, I started playing in late 2013 or early 2014.

 I learned a bunch from running @vestom changes to run the analyzer on unmatched decodes `-A -I 1` (IIRC).  Didn't tell me much about the RF environment, but more about the assumptions regarding how bitbuffers are assembled and how the decoders work.  I was going to write some stuff up for the group but never got around to it.

The lacrosse sensors were the first thing I worked on, which I think was in either late 2013 or early 2014.

pclov3r, 2017-06-18:
@rct thank you for your response that this is correct behavior and intended.

On that note any suggestions on how to suppress the duplicate output?

zuckschwerdt, 2017-06-18:
@pclov3r Buffer the last line and compare for duplicates. What consumer are you running? If it's simple pipes to e.g. log files the `uniq` command might help.

zuckschwerdt, 2017-06-18:
Side note, incidentally both my LaCrosse TX-style devices use 3 repeats.

[image]

closed 2017-07-06 by pclov3r