merbanan/rtl_433 #569: New device: GE Color Effects Christmas Lights Remote

lukecyca, 2017-06-18:
This is a remote that has two buttons, used to control christmas lights. Here's a photo of the whole system:
http://www.toddfun.com/blog/wp-content/uploads/2011/11/100_4515.jpg

I previously reverse engineered the receiver module here. Not sure if this is helpful.
https://github.com/lukecyca/G35RF

I have two of these remotes. I captured 4 button presses of each button on each remote. 
[ge-coloreffects-remote.zip](https://github.com/merbanan/rtl_433/files/1083222/ge-coloreffects-remote.zip)

I'm really fascinated by this project, and would like to learn how to make a module to decode these, but I think I need a bit of guidance. Could you look at these captures and make sure it's feasible and give me a few hints?

zuckschwerdt, 2017-06-18:
A button press seems to be 4 repeats of an manchester encoded FSK packet with 20 bits warm up, a desync, and about 27 bits. Try `FSK_PULSE_PCM` and `bitbuffer_manchester_decode()` with a half-bit length of 55us.

zuckschwerdt, 2017-06-18:
Start off by modifing `tpms_citroen.c` -- it should mostly fit.

zuckschwerdt, 2017-06-18:
Your data should be:
```
remote1-func1.data  {27} ea ab 6b 60 : 11101010 10101011 01101011 011
remote1-func2.data  {27} ea ab 6b 60 : 11101010 10101011 01101011 011
remote1-func3.data  {27} ea ab 6b 60 : 11101010 10101011 01101011 011
remote1-func4.data  {27} ea ab 6b 60 : 11101010 10101011 01101011 011
remote1-onoff1.data {27} ea ab 6d a0 : 11101010 10101011 01101101 101
remote1-onoff2.data {27} ea aa db 60 : 11101010 10101010 11011011 011
remote1-onoff3.data {27} ea ab 6d a0 : 11101010 10101011 01101101 101
remote1-onoff4.data {27} ea aa db 60 : 11101010 10101010 11011011 011
remote2-func1.data  {27} ea ab 6b 60 : 11101010 10101011 01101011 011
remote2-func2.data  {27} ea ab 6b 60 : 11101010 10101011 01101011 011
remote2-func3.data  {27} ea ab 6b 60 : 11101010 10101011 01101011 011
remote2-func4.data  {27} ea ab 6b 60 : 11101010 10101011 01101011 011
remote2-onoff1.data {27} ea aa db 60 : 11101010 10101010 11011011 011
remote2-onoff2.data {27} ea ab 6d a0 : 11101010 10101011 01101101 101
remote2-onoff3.data {27} ea aa db 60 : 11101010 10101010 11011011 011
remote2-onoff4.data {27} ea ab 6d a0 : 11101010 10101011 01101101 101
```

Could be inverted, i.e.:
```
remote1-func1.data  {27} 15 54 94 80 : 00010101 01010100 10010100 100
remote1-func2.data  {27} 15 54 94 80 : 00010101 01010100 10010100 100
remote1-func3.data  {27} 15 54 94 80 : 00010101 01010100 10010100 100
remote1-func4.data  {27} 15 54 94 80 : 00010101 01010100 10010100 100
remote1-onoff1.data {27} 15 54 92 40 : 00010101 01010100 10010010 010
remote1-onoff2.data {27} 15 55 24 80 : 00010101 01010101 00100100 100
remote1-onoff3.data {27} 15 54 92 40 : 00010101 01010100 10010010 010
remote1-onoff4.data {27} 15 55 24 80 : 00010101 01010101 00100100 100
remote2-func1.data  {27} 15 54 94 80 : 00010101 01010100 10010100 100
remote2-func2.data  {27} 15 54 94 80 : 00010101 01010100 10010100 100
remote2-func3.data  {27} 15 54 94 80 : 00010101 01010100 10010100 100
remote2-func4.data  {27} 15 54 94 80 : 00010101 01010100 10010100 100
remote2-onoff1.data {27} 15 55 24 80 : 00010101 01010101 00100100 100
remote2-onoff2.data {27} 15 54 92 40 : 00010101 01010100 10010010 010
remote2-onoff3.data {27} 15 55 24 80 : 00010101 01010101 00100100 100
remote2-onoff4.data {27} 15 54 92 40 : 00010101 01010100 10010010 010
```

Also first/last bit might be fixed and can be dropped, 24 bits would look more reasonable.

lukecyca, 2017-06-19:
Thanks @zuckschwerdt. I've created PR #570 for your review.

I found that it was not quite Manchester encoded, but the encoding is described in [this blog post](http://www.deepdarc.com/2010/11/27/hacking-christmas-lights/) so that's what I followed.

This code works pretty well on the captures I uploaded, and also running "live" and clicking the buttons on the remote. Pretty cool!

zuckschwerdt, 2017-06-19:
I highly doubt that encoding scheme is actually what's happening. The author of that blog post maybe isn't aware of manchester encoding. The description and timing given there does not seem to match. The dumps feature plain manchester, even the preamble is manchester encoded. FSK+Manchester is a typical pattern.

zuckschwerdt, 2017-06-19:
Looking closer, neither manchester or that custom scheme can be falsified with this data. The custom scheme shows a more plausible output, though.

lukecyca, 2017-06-19:
Pushed a new commit addressing your feedback - thanks!

With these changes, all of the example decode cleanly with all four repeats.

> Looking closer, neither manchester or that custom scheme can be falsified with this data. The custom scheme shows a more plausible output, though.

I agree that there is not enough example data to make a really strong case for either approach. With the custom encoding, a 1 uses twice as much space as a 0, but even that doesn't provide any clues because the three commands (0x55, 0x5a, 0xaa) all have equal quantities of 1s and 0s.

AFAICT, interpreting this with Manchester encoding would require throwing away one of the first bits in order to align the remaining bits into a valid Manchester-encoded string. This feels a bit arbitrary.

Shall we go with the custom encoding scheme?

zuckschwerdt, 2017-06-19:
Looks good to me. In the manchester decode every `1` is always followed by a `0`, which sure seems like an odd coincidence.

zuckschwerdt, 2017-12-03:
merged. thanks!

commit 07de3645 (2017-12-03): Device handler for GE Color Effects remote control
Fixes #569
closed 2017-12-03 by zuckschwerdt