merbanan/rtl_433 #571: JSON Output Examples?

efaden, 2017-06-27:
Is there a way to get the program to dump out what the JSON output for the different formats will look like?  I am trying to write an importer to convert JSON to prometheus.io for graphing, but I need a template I can parse.  Thanks.

zuckschwerdt, 2017-06-27:
JSON output will be a simple (flat) object with key-value pairs. Use `-F csv` and you'll get a listing of all the possible keys. Value types will be string or number only.

rct, 2017-06-27:
The rtl_433_tests repo might be of help to you:

* [Browse the repo online](https://github.com/merbanan/rtl_433_tests/tree/master/tests/) - Look for .json files for examples.  This is captured output that is used for comparison during testing.  
* Or, clone the repo to use the all of the recorded signals to generate your output with `rtl_433 -q -r foo.data`

closed 2018-01-28 by zuckschwerdt

geekdave, 2018-11-03:
@efaden did you ever find a solution for exporting these metrics to prometheus?  I'm looking to do the same.

zuckschwerdt, 2018-11-03:
There is an outline at http://triq.net/articles/rtl-433-networking, specifically the [Graphite relay](https://github.com/zuckschwerdt/rtl_433_relay/blob/master/rtl_433_graphite_relay.py) should be close to what you need.