merbanan/rtl_433 #578: Do you need transmitter or base station to work with door sensor (e.g. KERUI) or PIR motion, or can you just passively monitor it?

aindilis, 2017-07-10:
Hi,

I saw this codebase and decided to order a $3 door sensor (seems a knockoff of a KERUI D025).  I am wondering if anyone could help me out and tell me if I can just run this rtl-433 code on a DVB-T (which seems to work so far) when the device arrives from China in a month and just intercept a passive periodic status update, or if it works differently (i.e. you have to query it with an rf transmission, for instance, or you have to "pair" it or something with a base station).  If I do require transmission ability, is there a reference to how to set this up on the cheap (including ordering different hardware that is better supported).  Like I saw some kind of microcontroller based transmitter circuit for ~ $5 on ebay.  This is both for me and for people experiencing poverty using the application I've been developing here: 

http://facebook.com/frdcsa 

http://frdcsa.org/~andrewdo/WebWiki/FreeLifePlanningCoachSoftwareUpdate.html  

I'd appreciate some hand holding, as I can then relay this instruction to others, because this freelifeplanner.org application (when finished) has an "AI" planner than can help promulgate itself to people in need via observation, dialog and prompting with its users and their social network.

Best,
Andrew

zuckschwerdt, 2017-07-10:
Cheap magnetic door sensors will usually send a single burst of packets when the trigger condition is detected, i.e. the door is opened. There won't be periodic updates or receiver (querying) capabilities.
If the sensor data isn't decoded properly open an issue with some sample data (`-A` mode, `-a` mode, and `-a -t` files) and I'll help out.

aindilis, 2017-07-10:
Awesome, thanks for the help :)  I'll follow up when the sensors arrive and I test them out.

merbanan, 2017-07-10:
@aindilis can you post a link to a device you bought?

aindilis, 2017-07-10:
KERUI (possible imitation?) Door/Window Sensor

http://www.gearbest.com/alarm-systems/pp_659761.html

Generic PIR Motion

http://www.ebay.com/itm/Wireless-PIR-Infrared-IR-Motion-Auto-Sensor-Alarm-Detector-Home-Security-System/112385758894?ssPageName=STRK%3AMEBIDX%3AIT&var=412791942156&_trksid=p2057872.m2749.l2649

I cannot find the URL for original page that I bought the RTL-SDR:

0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T

uses rtl8192cu module.

aindilis, 2017-07-25:
Update: (no reply solicited) the PIR Sensor arrived and seems to work with -A 87.  It definitely outputs a repeated sequence of something in bursts separated by maybe a few seconds when it is turned on and there is motion, and otherwise nothing.  Although I don't understand the output and whether it contains any information such as the amount of motion detected or something, or if it's in the timing patterns.  Writing an Expect wrapper with Perl to intercept stdout and send a message to the Free Life Planner when it detects such movement.  Works from the 2nd floor through to the 1st floor.  Don't know whether there is a way to uniquely identify multiple PIR sensors (there has to be, right?) and/or whether there is a way to listen to different types of sensors simultaneously (assuming of course there is).  Going to reorder as many of these $4 generic PIR wireless sensors as needed for our household.  Will report on the door sensor when that arrives.  Thanks again for helping me along, this has been a great help to me, my family and my project.  And of course thanks for writing such an awesome piece of software!  :)

zuckschwerdt, 2017-07-25:
It would be best if you capture a sample (`-a -t`) upload it somewhere and let me know.

aindilis, 2017-07-31:
Ah, srry somehow missed this - I will do so by end of day - thank you!!!

zuckschwerdt, 2018-02-17:
for anyone following: doorbell-style sensors are now best handled by the flex decoder, see e.g. #658 for a discussion.

closed 2018-02-17 by zuckschwerdt