merbanan/rtl_433 #579: Acurite tower sensor stopped working

kk7ds, 2017-07-10:
I've been running 922a76a81820c68e2e5c55a428e5d22135a8be9a for over a year with good success. I went to update to 1be0118aa74dd2e9d151a0dcb72a2c78d505539d today and now I don't get any data from my "Acurite tower sensor" devices. I don't get errors or anything, I just never see them. I get data from Oregon Scientific and Lacrosse sensors, but just never hear from the Acurites. If I switch back to the old code, they come right in no problem.

I'm not sure what to do to further debug this, but I'm happy to help if you let me know what you need.

zuckschwerdt, 2017-07-10:
The "Acurite tower sensor" is handled by the "Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning" device. Sample data is in `rtl_433_tests/tests/acurite/03/`, I guess. Can you check if that matches your device and problem? Otherwise I'd need sample data.

kk7ds, 2017-07-10:
I'm not getting any acurite data (even as a different name) so I assume it's not matching.

What do I need to do to get you sample data?

kk7ds, 2017-07-10:
Oh jeez. I was looking at the -h output to see how to collect data and I see that my sensor is now disabled by default. With -G I seem to get the data no problem. Sorry for the noise.

closed 2017-07-10 by kk7ds

rct, 2017-07-10:
Hey Dan, @kk7ds - Did you enabling the Acurite output with either `-G` or -`-R 40`?

Note the output format will have changed between versions as the tower and 5-n-1 both output structured data now (JSON, KV, CSV).    If you don't enable one of those the text strings will still be somewhat different.

Another change in the last year, is that decoders that don't emit structured data aren't enabled by default.  Currently the Acurite protocol for the tower, 5-n-1, and lightning detector aren't enabled.  This is my fault.  The lightning detector hasn't been converted over to use structured/JSON output yet.  I need to get back to that.

Hope this helps and 73, w2rct

kk7ds, 2017-07-10:
Yep, as above, I had missed the -G thing. The commit I was working on before didn't support JSON for all of my sensors, but I think they all work now, so I'm stoked to be able to use that instead of scraping the regular text output for my use :)